RobotWebTools/roslibpy #2: Python 2.7 Service

mrlowalowa, 2018-07-04:
Hello,

thank you for providing such a rich featured python library for the ROSBridge interface.
I started using ROSBridge myself in the last few months and created a minimal library for ROSBridge in python2.7 with ws4py. Right now it does only support the basic ROS functionalities of actively advertising and publishing messages. For further use cases the full stack needs to be implemented like you did it with your roslibpy library.

## Expected Result
I wanted to test the connectivity of your library with my server and list all available topics.

## Actual Results

Only this error code gets generated.

>  2018-07-04 05:12:58,269 [DEBUG] Server connected: tcp4:192.168.1.225:8080
2018-07-04 05:12:58,278 [INFO] Connection to ROS MASTER ready.
2018-07-04 05:12:58,285 [DEBUG] Sending ROS service request: {"args": {}, "id": "call_service:/rosapi/topics:1", "service": "/rosapi/topics", "op": "call_service"}
Unhandled Error
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/twisted/python/log.py", line 103, in callWithLogger
    return callWithContext({"system": lp}, func, *args, **kw)
  File "/usr/lib/python2.7/site-packages/twisted/python/log.py", line 86, in callWithContext
    return context.call({ILogContext: newCtx}, func, *args, **kw)
  File "/usr/lib/python2.7/site-packages/twisted/python/context.py", line 122, in callWithContext
    return self.currentContext().callWithContext(ctx, func, *args, **kw)
  File "/usr/lib/python2.7/site-packages/twisted/python/context.py", line 85, in callWithContext
    return func(*args,**kw)
--- <exception caught here> ---
  File "/usr/lib/python2.7/site-packages/twisted/internet/posixbase.py", line 614, in _doReadOrWrite
    why = selectable.doRead()
  File "/usr/lib/python2.7/site-packages/twisted/internet/tcp.py", line 205, in doRead
    return self._dataReceived(data)
  File "/usr/lib/python2.7/site-packages/twisted/internet/tcp.py", line 211, in _dataReceived
    rval = self.protocol.dataReceived(data)
  File "/usr/lib/python2.7/site-packages/autobahn/twisted/websocket.py", line 131, in dataReceived
    self._dataReceived(data)
  File "/usr/lib/python2.7/site-packages/autobahn/websocket/protocol.py", line 1175, in _dataReceived
    self.consumeData()
  File "/usr/lib/python2.7/site-packages/autobahn/websocket/protocol.py", line 1187, in consumeData
    while self.processData() and self.state != WebSocketProtocol.STATE_CLOSED:
  File "/usr/lib/python2.7/site-packages/autobahn/websocket/protocol.py", line 1553, in processData
    fr = self.onFrameEnd()
  File "/usr/lib/python2.7/site-packages/autobahn/websocket/protocol.py", line 1675, in onFrameEnd
    self._onMessageEnd()
  File "/usr/lib/python2.7/site-packages/autobahn/twisted/websocket.py", line 159, in _onMessageEnd
    self.onMessageEnd()
  File "/usr/lib/python2.7/site-packages/autobahn/websocket/protocol.py", line 627, in onMessageEnd
    self._onMessage(payload, self.message_is_binary)
  File "/usr/lib/python2.7/site-packages/autobahn/twisted/websocket.py", line 162, in _onMessage
    self.onMessage(payload, isBinary)
  File "/usr/lib/python2.7/site-packages/roslibpy/comm.py", line 96, in onMessage
    handler(message)
  File "/usr/lib/python2.7/site-packages/roslibpy/comm.py", line 117, in _handle_service_response
    callback(ServiceResponse(message['values']))
exceptions.TypeError: 'str' object is not callable
2018-07-04 05:12:58,488 [CRITICAL] Unhandled Error
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/twisted/python/log.py", line 103, in callWithLogger
    return callWithContext({"system": lp}, func, *args, **kw)
  File "/usr/lib/python2.7/site-packages/twisted/python/log.py", line 86, in callWithContext
    return context.call({ILogContext: newCtx}, func, *args, **kw)
  File "/usr/lib/python2.7/site-packages/twisted/python/context.py", line 122, in callWithContext
    return self.currentContext().callWithContext(ctx, func, *args, **kw)
  File "/usr/lib/python2.7/site-packages/twisted/python/context.py", line 85, in callWithContext
    return func(*args,**kw)
--- <exception caught here> ---
  File "/usr/lib/python2.7/site-packages/twisted/internet/posixbase.py", line 614, in _doReadOrWrite
    why = selectable.doRead()
  File "/usr/lib/python2.7/site-packages/twisted/internet/tcp.py", line 205, in doRead
    return self._dataReceived(data)
  File "/usr/lib/python2.7/site-packages/twisted/internet/tcp.py", line 211, in _dataReceived
    rval = self.protocol.dataReceived(data)
  File "/usr/lib/python2.7/site-packages/autobahn/twisted/websocket.py", line 131, in dataReceived
    self._dataReceived(data)
  File "/usr/lib/python2.7/site-packages/autobahn/websocket/protocol.py", line 1175, in _dataReceived
    self.consumeData()
  File "/usr/lib/python2.7/site-packages/autobahn/websocket/protocol.py", line 1187, in consumeData
    while self.processData() and self.state != WebSocketProtocol.STATE_CLOSED:
  File "/usr/lib/python2.7/site-packages/autobahn/websocket/protocol.py", line 1553, in processData
    fr = self.onFrameEnd()
  File "/usr/lib/python2.7/site-packages/autobahn/websocket/protocol.py", line 1675, in onFrameEnd
    self._onMessageEnd()
  File "/usr/lib/python2.7/site-packages/autobahn/twisted/websocket.py", line 159, in _onMessageEnd
    self.onMessageEnd()
  File "/usr/lib/python2.7/site-packages/autobahn/websocket/protocol.py", line 627, in onMessageEnd
    self._onMessage(payload, self.message_is_binary)
  File "/usr/lib/python2.7/site-packages/autobahn/twisted/websocket.py", line 162, in _onMessage
    self.onMessage(payload, isBinary)
  File "/usr/lib/python2.7/site-packages/roslibpy/comm.py", line 96, in onMessage
    handler(message)
  File "/usr/lib/python2.7/site-packages/roslibpy/comm.py", line 117, in _handle_service_response
    callback(ServiceResponse(message['values']))
exceptions.TypeError: 'str' object is not callable
2018-07-04 05:12:58,501 [INFO] WebSocket connection closed: connection was closed uncleanly (peer dropped the TCP connection without previous WebSocket closing handshake)
2018-07-04 05:12:58,506 [DEBUG] Lost connection. Reason: [Failure instance: Traceback (failure with no frames): <type 'exceptions.TypeError'>: 'str' object is not callable
]
2018-07-04 05:12:58,516 [INFO] <twisted.internet.tcp.Connector instance at 0xb69b79cc> will retry in 2 seconds
2018-07-04 05:12:58,525 [INFO] Stopping factory <roslibpy.comm.RosBridgeClientFactory object at 0xb71d214c>
2018-07-04 05:13:00,740 [INFO] Starting factory <roslibpy.comm.RosBridgeClientFactory object at 0xb71d214c>
2018-07-04 05:13:00,750 [DEBUG] Started to connect...
2018-07-04 05:13:01,019 [DEBUG] Server connected: tcp4:192.168.1.225:8080
2018-07-04 05:13:01,024 [INFO] Connection to ROS MASTER ready.
2018-07-04 05:13:01,127 [INFO] WebSocket connection closed: None
2018-07-04 05:13:01,133 [DEBUG] Lost connection. Reason: [Failure instance: Traceback (failure with no frames): <class 'twisted.internet.error.ConnectionDone'>: Connection was closed cleanly.
]
2018-07-04 05:13:01,142 [INFO] <twisted.internet.tcp.Connector instance at 0xb69b79cc> will retry in 6 seconds
2018-07-04 05:13:01,151 [INFO] Stopping factory <roslibpy.comm.RosBridgeClientFactory object at 0xb71d214c>

When I print the message['values'] I get an Json string encoded with utf8.
The values json string contained already all the information I needed but the library was not able to present me the results.
Naturally I tried to fix this with altering the data type of message['values'] but the furthest I came was an error message telling me that the update() function in core.py needs 2 args but got only one.
Maybe this is only a python2.7 language error. I hope you can help me here! :) 
## Reproduction Steps
```python
import roslibpy

ros = roslibpy.Ros("192.168.1.225", "8080")
import logging

FORMAT = '%(asctime)-15s [%(levelname)s] %(message)s'
logging.basicConfig(level=logging.DEBUG, format=FORMAT)

ros.on_ready(lambda: ros.get_topics("print"))
ros.call_later(3, ros.close)
ros.call_later(5, ros.terminate)

ros.run_event_loop()

```

## System Information
IoT2000 - Yocto Linux, Python2.7, ROS kinetic (on different machine), recent ROSBridge version

mrlowalowa, 2018-07-04:
I found the solution.. I was trying to get away with the syntax error of the callback function print by making a string out of it. That was the issue.. "str is not callable"

The original problem was that python2.7 does not have print as a function. This must be included from __future__

this is the working code..:

```python
from __future__ import print_function
import roslibpy

ros = roslibpy.Ros("192.168.1.225", "8080")
import logging

FORMAT = '%(asctime)-15s [%(levelname)s] %(message)s'
logging.basicConfig(level=logging.DEBUG, format=FORMAT)

ros.on_ready(lambda: ros.get_topics(print))
ros.call_later(3, ros.close)
ros.call_later(5, ros.terminate)

ros.run_event_loop()
```

closed 2018-07-04 by mrlowalowa

gonzalocasas, 2018-07-04:
Aha! Ok, cool to hear that you solved it 👍

mrlowalowa, 2018-07-04:
Took me a night and a bit of mate but was a great learning.. ;)

I would recommend adding these examples for implementing your library in different projects. Test code inside the library is also a great thing but dangerous as we can see here..
Better capsulate the whole example for an easy to use in all environments :)

gonzalocasas, 2018-07-04:
This mate [image]

or this other one

[image]
? 

😄 

But yes, you're right, I'll add a few examples to the docs to show more clearly how to use it.
Thanks for reporting (in great detail) btw!

mrlowalowa, 2018-07-04:
The first one of course :+1: 
:robot: robots need :rocket: :fuelpump: 

Right now I'm rebuilding the sensor_msgs/IMU because I want to use this lib to publish IMU data to my ROS master. Do you have any recommendations where I can get the complete ros_msgs stack without an installed ROS distro?

Maybe this is a point that can be added for the future. Sending the data through ros-strings and rebuilding them inside the ROS-Master is probably not very effective.

At least I can report my findings and improve this lib when I'll use it for my projects! :wink: 

Greetings from Bavaria..

gavanderhoorn, 2018-07-05:
@mrlowalowa wrote:
> Do you have any recommendations where I can get the complete ros_msgs stack without an installed ROS distro?

I'm not sure I understand. I don't know of any package called `ros_msg`, or a set of packages that goes by that name.

gonzalocasas, 2018-07-05:
Maybe I'm assuming too much, but I guess @mrlowalowa means the whole of `std_msgs`, `common_msgs`, `industrial_msgs`, and everything in between (I know that's not very precise either ;)). Is that what you meant?

mrlowalowa, 2018-07-05:
Hello everyone,

yes you correctly interpreted my question.
I want to use this library to publish sensor data to my master. This data will be used by other (standard) ros packages like robot_localization. Therefore it is useful to publish the data in the needed format.
In the end all these msgs types are bundled in a json and these string bits are interpreted as float64 etc etc.
Sometimes it is very easy to create these msgs in python via dictionaries (i.e. sending a Bool). But it would be quite useful to provide these dictionaries for more complex msgs  and only fill your data into the right spots.

Here is a description of some of these msgs:
http://wiki.ros.org/sensor_msgs (this is part of the already mentioned `common_msgs` package)

When we take the `sensor_msgs/IMU` msg for example. It consists of many different subtypes that I need to build by my own. And even the documentation is not 100% clear for me how the header part with the time stamp was build. Gladly I could analyze a different raw msg from my robot.

Normally this is not a problem when you are in a ROS environment. There you only need this:

```
import rospy
from sensor_msgs.msg import Imu, Temperature, MagneticField
```

And you are all set. Take this code here for example where the msgs is filled with data from the IMU I want to include:
https://github.com/mdrwiega/bosch_imu_driver/blob/master/nodes/bosch_imu_node.py#L232

The only problem here is that these sensor_msgs depend on a installed ROS distro for filling i.e. the header with the time stamp via rospy.Time.now() and for the inside structure. I found no way to include them right away from the ROS github. A standalone library with the msgs would be very useful out of my opinion.

gavanderhoorn, 2018-07-05:
Perhaps @gonzalocasas (or any of the other developers) can clarify how `roslibpy` code is supposed to interact with msgs on the client side. I haven't used it myself so I cannot suggest anything.

> Therefore it is useful to publish the data in the needed format.

no, it's actually a requirement. And `rosbridge` will only work correctly if you do.

If there is no 'message' object type in `roslibpy`, then creating dictionaries with the appropriate / expected fields seems like your only option.

gavanderhoorn, 2018-07-05:
If you're just looking for an exhaustive dump of all the fields and constants in all the messages in ROS (which is difficult as it's unclear what 'all' means here), perhaps something like:

```
rosmsg list | rosmsg show
```

could help.

That would at least help understand *which* fields are in certain messages.

It won't make it any easier to populate them from your Python code though. That would be something the `roslibpy` devs could assist with (some helper lib?).

mrlowalowa, 2018-07-05:
> > 
> 
>     
> > Therefore it is useful to publish the data in the needed format.
> 
> no, it's actually a requirement. And rosbridge will only work correctly if you do.

yes it is a requirement to use payload that comply with any ros message type (suitable to your topic).
But it does not require you to use the best suitable message type. Another way is to publish the whole thing in a own data structure, pack it in a string and write a ros node that converts that string /data structure to the wished message type.
My problem here is that I want to use the data for localization what is a time critical activity and this work around would not be very efficient.

gavanderhoorn, 2018-07-05:
> Another way is to publish the whole thing in a own data structure, pack it in a string and write a ros node that converts that string /data structure to the wished message type.

if you feel that is acceptable then I don't understand why you would want to use `rosbridge`.

mrlowalowa, 2018-07-05:
no it is not acceptable for me. I just wanted to show the different possibilities to solve this issue.
The java rosbridge client seems to implement a subset of ros messages.

gavanderhoorn, 2018-07-05:
> The java rosbridge client seems to implement a subset of ros messages.

that's sort of what I suggested, but will have to be explicitly created and added. Either by you (ie: a user), or by the `roslibpy` devs.

It could be easily automated though. Using the output of `rosmsg show` to drive a generator script.

gonzalocasas, 2018-07-05:
> > If there is no 'message' object type in roslibpy, then creating dictionaries with the appropriate / expected fields seems like your only option.

> The java rosbridge client seems to implement a subset of ros messages.

In Java, it is idiomatic to have strongly-typed messages, in Python it would be awkward to do so, hence dictionaries that comply to the payload is the way to go. 

However, having a generator script as @gavanderhoorn suggests, would be an interesting addition, perhaps as part of the `pyinvoke` tasks in the lib.

> My problem here is that I want to use the data for localization what is a time critical activity and this work around would not be very efficient.

This part I didn't fully get. Do you mean you want to be in control of filling in the header of the message, only to use the timestamp to do localization (as in trilateration/time-of-arrival/similar approach)? If that's the case, then I'm not sure if the accuracy of the entire stack is good enough, in particular, python itself doesn't give you much time guarantees so. I guess it depends a lot on your setup, your requirements and the expected accuracy you want to achieve, but sounds like a tricky proposition ;)

mrlowalowa, 2018-07-05:
exactly. I was thinking about the same and my main goal for now was to ask if there are already independent libraries offering this feature that are used in combination with `roslibpy`.

> It could be easily automated though. Using the output of `rosmsg show` to drive a generator script.

That seems like a fast and promising solution. This command is currently not working for me as my ros master is out of reach. But I'll look into it later.
@gonzalocasas what is the plan for roslibpy? Are any further developments planed?

gavanderhoorn, 2018-07-05:
@mrlowalowa wrote:
> This command is currently not working for me as my ros master is out of reach.

`rosmg` does not need a ros master.

gonzalocasas, 2018-07-05:
> @gonzalocasas what is the plan for `roslibpy`? Are any further developments planned?

yes, the immediate next step is to add a communication layer for `IronPython` to be able to use it from within Rhino/Grasshopper (this is almost ready, to be pushed in the coming days), and a bit longer term, this library is to become a core component of the [COMPAS framework](https://compas-dev.github.io/).

ipa-mdl, 2018-07-05:
IMHO `roslibpy` is meant for environments without access to the ROS message definitions (and the generated python files). If you have access to them, you could use `rospy` instead.
(If you can reach the rosbridge, you should be able to reach the master).

Of course there is always room for convenience helpers.
However, I don't see the need for generated classes in Python.

I am creating objects on-the-fly like this:
```
def wrap_result(name, res):
    return type(name, (object,), res)()

 wrap_result("my_pkg/MyMsg", msg.data)
```

mrlowalowa, 2018-07-05:
> Do you mean you want to be in control of filling in the header of the message, only to use the timestamp to do localization

No? I need the msgs frame and already fill this with the time stamp information. It was tricky for me to reverse the exact structure of this header.

Optimally I only wish to fill a ready to use dict and send it via `roslibpy` to my master.
The localization will be based on imu, gps, laserscanner. As I want to use a kalmanfilter I suggest that it will take the timestamp into its computing to generate a position. But this is a topic I still have to look into.
First thing for me is it to publish the data in a standard format to use it with other ROS nodes.
I have to see if the python & network combination will work for my localization and if going to C/C++ would fix this.

Right now this is my approach to generate a header with the actual time stamp:

```python
import time
#import bno055

ros_header = {
	"header": {
		"seq" : 0,
		"stamp" : {
			"secs" : 0,
			"nsecs": 0
		},
		"frame_id": "world" #probably imu_link
	}
}
print (ros_header)
time = ("%.9f" % time.time()).split('.')
print(time)
ros_header['header']['stamp']['secs'] = time[0]
ros_header['header']['stamp']['nsecs'] = time[1]
print (ros_header)
```

mrlowalowa, 2018-07-05:
> IMHO `roslibpy` is meant for environments without access to the ROS message definitions (and the generated python files)

This is exactly my usecase and also my understanding of the need of `rosbridge`.

ipa-mdl, 2018-07-05:
> Optimally I only wish to fill a ready to use dict and send it via roslibpy to my master.

This is the current API. Can you provide an example that shows the API you want to use?

mrlowalowa, 2018-07-05:
@ipa-mdl thanks for taking part in the conversation.
> This is the current API.
Which API are you referring to? The general ROS API or something that is accessible for me only via this roslibpy library?

I already mentioned the code I want to use with my BNO055 IMU sensor.
> https://github.com/mdrwiega/bosch_imu_driver/blob/master/nodes/bosch_imu_node.py#L232
Is this a suitable API example for you?

Am I wrong that roslibpy takes any kind of string, force it to be a dict and sends it. See here:
https://github.com/gramaziokohler/roslibpy/blob/master/src/roslibpy/comm.py#L34

gonzalocasas, 2018-07-05:
> Am I wrong that roslibpy takes any kind of string, force it to be a dict and sends it. 

That's *almost* exactly correct. `Message` inherits from `UserDict`, so the cast to `dict` is a technicality to make the `json` encoder work.

More generally, the idea of "send any dict over the wire" is not a decision we took at `roslibpy` level, but it's inherited of [roslibjs](https://github.com/RobotWebTools/roslibjs) (by the authors of `rosbridge`), after which this library is closely modelled.

mrlowalowa, 2018-07-05:
thank you for making it clearer :)

But I still do not know what @ipa-mdl refers to as API

I know that rosbridge "accepts" everything as incoming message which is then rejected by the ROS system if it does not comply with the message type of the corresponding topic. 
Out if my opinion it is a good thing to not have the same functionality twice and therefore providing a lower overhead.
I'm just wishing for a nice "message_type" library that makes the usage of `rosbridge` as smooth as the normal usage with `rospy`  

This is now my `sensor_msgs/Imu` for anybody who wants to use it and my kind of way how I would build up such a "message_type" library:

```python
import time

ros_header = {
	"seq" : 0,
	"stamp" : {
		"secs" : 0,
		"nsecs": 0
	},
	"frame_id": "world" #probably imu_link
}

ros_geo_quaternion = {
	"x" : 0,
	"y" : 0,
	"z" : 0,
	"w" : 0,
}

ros_geo_vector3 = {
	"x" : 0,
	"y" : 0,
	"z" : 0,
}

covariance = [0] * 9

ros_sensor_imu = {}
ros_sensor_imu["header"] = ros_header.copy()
ros_sensor_imu["orientation"] = ros_geo_quaternion.copy()
ros_sensor_imu["orientation_covariance"] = covariance[:]
ros_sensor_imu["angular_velocity"] = ros_geo_vector3.copy()
ros_sensor_imu["angular_velocity_covariance"] = covariance[:]
ros_sensor_imu["linear_acceleration"] = ros_geo_vector3.copy()
ros_sensor_imu["linear_acceleration_covariance"] = covariance[:]

print (ros_sensor_imu)

time = ("%.9f" % time.time()).split('.')
print(time)
ros_sensor_imu['header']['stamp']['secs'] = time[0]
ros_sensor_imu['header']['stamp']['nsecs'] = time[1]
print (ros_header)

```

The timing part for the header is also a wish that should be included at least in examples of how to use the types or the other way around: how to rebuild rospy functions.
Maybe something for another library.

gonzalocasas, 2018-07-05:
@mrlowalowa the behavior on the `header` is actually a pretty special case. It's mentioned in the `ROSBRIGE_PROTOCOL` spec: 

https://github.com/RobotWebTools/rosbridge_suite/blob/f0d359797eb90b322e5eb23cfda89ea7a9aceac8/ROSBRIDGE_PROTOCOL.md#343-publish--publish-

> Special case: if the type being published has a 'header' field, then the client can optionally omit the header from the msg. If this happens, rosbridge will automatically populate the header with a frame id of "" and the timestamp as the current time. Alternatively, just the timestamp field can be omitted, and then the current time will be automatically inserted.

I will try to add a few examples to the docs showing this and some other use cases.

mrlowalowa, 2018-07-05:
Oh yes! Thank you for that! Now you start answering questions before I ask them.. :smile: 

http://wiki.ros.org/roslibjs/Tutorials/Publishing%20video%20and%20IMU%20data%20with%20roslibjs#CA-e105c59ec3a3c3894889cbce51f2ebb90a5cf6c0_172

Only the frame_id was overwritten..
Thank you for making that clear to me!

Another question I have..
I want to publish my data in a loop running as long as the robot is on.
My plan: write a function that publishes the data from inside a `while True:` loop and start it via:
```python
ros_client.on_ready(start_sending, run_in_thread=True)
ros_client.run_event_loop()
```
Now I started to see that `while True:` is not ideal as your lib relays on `Reactor`.
https://stackoverflow.com/questions/35111265/how-does-pythons-twisted-reactor-work#35119594
Do you have any recommendation to do it the proper way?

This would also be a quite useful example.. :grin:

gonzalocasas, 2018-07-05:
> Now I started to see that `while True:` is not ideal as your lib relays on `Reactor`.

Do you mean you want a function that outlives the `Reactor` and is linked to your robot's lifetime? Usually the reactor runs for the lifetime of your full program (since reactors cannot be restarted), so, it would still be ok to use that `while`, since that's basically the maximum lifespan your code will get anyway.

But perhaps I'm missing the point. Could you maybe clarify a bit further what you mean?

mrlowalowa, 2018-07-05:
I would just not issue the `ros.close()` and `ros.terminate()` so the programm runs nearly for ever.

I was just a but unsure about the usage as reactor alters the functionality of `while True:` but that does not hinders me to use it. This whole question was based on the second link I posted.

I was just searching for the right way to interact with reactor but I guess I just should not have started thinking about the underlying framework and used the normal `while` loop.

gonzalocasas, 2018-07-05:
I wouldn't say the `reactor` alters the `while`, it simply has the power to interrupt it if the main event loop is terminated.

And as you say, don't focus too much on the underlying framework, especially since
 1) when running on IronPython, I am not relying on it at all, since reactors don't work there
 2) being an implementation detail, it's subject to be replaced at some point in the future if we see fit (and it's something I've considered a couple of times already because these twisted reactors are highly opinionated).