RobotWebTools/roslibpy #5: Unable to connect to master

Behery, 2018-07-30:
Hi,
I'm unable to connect to rosmaster (running on the same machine). Maybe i'm missing something

## Expected Result
`ros.is_connected = True`

## Actual Result
`ros.is_connected = True`

## Reproduction Steps
```python
## The client
import roslibpy

ros = roslibpy.Ros(host="localhost", port="9090")
ros.connect()
ros.is_connected
```
### Output on the backend:
``` 
SUMMARY
========

PARAMETERS
 * /rosapi/params_glob: [*]
 * /rosapi/services_glob: [*]
 * /rosapi/topics_glob: [*]
 * /rosbridge_websocket/address: 
 * /rosbridge_websocket/authenticate: False
 * /rosbridge_websocket/bson_only_mode: False
 * /rosbridge_websocket/delay_between_messages: 0
 * /rosbridge_websocket/fragment_timeout: 600
 * /rosbridge_websocket/max_message_size: None
 * /rosbridge_websocket/params_glob: [*]
 * /rosbridge_websocket/port: 9090
 * /rosbridge_websocket/retry_startup_delay: 5
 * /rosbridge_websocket/services_glob: [*]
 * /rosbridge_websocket/topics_glob: [*]
 * /rosbridge_websocket/unregister_timeout: 10
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /
    rosapi (rosapi/rosapi_node)
    rosbridge_websocket (rosbridge_server/rosbridge_websocket)

auto-starting new master
process[master]: started with pid [3004]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to fa345d3a-93e3-11e8-88c9-08d40cdfa665
process[rosout-1]: started with pid [3017]
started core service [/rosout]
process[rosbridge_websocket-2]: started with pid [3028]
process[rosapi-3]: started with pid [3035]
registered capabilities (classes):
 - rosbridge_library.capabilities.call_service.CallService
 - rosbridge_library.capabilities.advertise.Advertise
 - rosbridge_library.capabilities.publish.Publish
 - rosbridge_library.capabilities.subscribe.Subscribe
 - <class 'rosbridge_library.capabilities.defragmentation.Defragment'>
 - rosbridge_library.capabilities.advertise_service.AdvertiseService
 - rosbridge_library.capabilities.service_response.ServiceResponse
 - rosbridge_library.capabilities.unadvertise_service.UnadvertiseService
[INFO] [1532946800.761421]: Rosbridge WebSocket server started on port 9090

```

## System Information
I'm on ubuntu 16.04 using ros kinetic

Behery, 2018-07-30:
I was setting the port as an integer, but put it here (in the issue) as a str by mistake.
For the rest you're correct. It worked when I used the `on_ready` handler with the `run_even_loop` (there was no `run_forever`; maybe i'm using a different version)

Thanks @gonzalocasas :) :)

closed 2018-07-30 by Behery

gonzalocasas, 2018-07-30:
@Behery ah! yes, we're just releasing `0.2.0` as we speak, and one of the changes is that `run_even_loop` is now deprecated in favor of `run_forever` ;)