RobotWebTools/roslibpy #6: TFClient breaks on callback in python3

Behery, 2018-07-31:
I'm trying to subscribe to a TFClient in python3 but I'm getting a `CRITICAL:twisted:Unhandled Error`
This is only happening in python3, but in python 2.7 it's working with the expected result below.

## Expected Result
{'rotation': {u'y': 0.0, u'x': 0.0, u'z': 0.0, u'w': 0.0}, 'translation': {u'y': 0.0, u'x': 0.0, u'z': 0.0}}.

## Actual Result
```python 
started
Unhandled Error
Traceback (most recent call last):
  File "/home/anwar/.local/lib/python3.5/site-packages/roslibpy/ros.py", line 62, in run_forever
    self.factory.manager.run_forever()
  File "/home/anwar/.local/lib/python3.5/site-packages/roslibpy/comm/comm_autobahn.py", line 128, in run_forever
    reactor.run()
  File "/home/anwar/.local/lib/python3.5/site-packages/twisted/internet/base.py", line 1261, in run
    self.mainLoop()
  File "/home/anwar/.local/lib/python3.5/site-packages/twisted/internet/base.py", line 1270, in mainLoop
    self.runUntilCurrent()
--- <exception caught here> ---
  File "/home/anwar/.local/lib/python3.5/site-packages/twisted/internet/base.py", line 896, in runUntilCurrent
    call.func(*call.args, **call.kw)
  File "/home/anwar/.local/lib/python3.5/site-packages/roslibpy/tf.py", line 95, in update_goal
    request, self._process_response, self._process_error)
  File "/home/anwar/.local/lib/python3.5/site-packages/roslibpy/core.py", line 248, in call
    }), callback, errback)
  File "/home/anwar/.local/lib/python3.5/site-packages/roslibpy/ros.py", line 143, in send_service_request
    self.factory.on_ready(_send_internal)
  File "/home/anwar/.local/lib/python3.5/site-packages/roslibpy/comm/comm_autobahn.py", line 74, in on_ready
    callback(self._proto)
  File "/home/anwar/.local/lib/python3.5/site-packages/roslibpy/ros.py", line 140, in _send_internal
    proto.send_ros_service_request(message, callback, errback)
  File "/home/anwar/.local/lib/python3.5/site-packages/roslibpy/comm/comm.py", line 79, in send_ros_service_request
    json_message = json.dumps(dict(message)).encode('utf8')
  File "/usr/lib/python3.5/json/__init__.py", line 230, in dumps
    return _default_encoder.encode(obj)
  File "/usr/lib/python3.5/json/encoder.py", line 198, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/usr/lib/python3.5/json/encoder.py", line 256, in iterencode
    return _iterencode(o, 0)
  File "/usr/lib/python3.5/json/encoder.py", line 179, in default
    raise TypeError(repr(o) + " is not JSON serializable")
builtins.TypeError: dict_keys(['turtle1']) is not JSON serializable

CRITICAL:twisted:Unhandled Error
Traceback (most recent call last):
  File "/home/anwar/.local/lib/python3.5/site-packages/roslibpy/ros.py", line 62, in run_forever
    self.factory.manager.run_forever()
  File "/home/anwar/.local/lib/python3.5/site-packages/roslibpy/comm/comm_autobahn.py", line 128, in run_forever
    reactor.run()
  File "/home/anwar/.local/lib/python3.5/site-packages/twisted/internet/base.py", line 1261, in run
    self.mainLoop()
  File "/home/anwar/.local/lib/python3.5/site-packages/twisted/internet/base.py", line 1270, in mainLoop
    self.runUntilCurrent()
--- <exception caught here> ---
  File "/home/anwar/.local/lib/python3.5/site-packages/twisted/internet/base.py", line 896, in runUntilCurrent
    call.func(*call.args, **call.kw)
  File "/home/anwar/.local/lib/python3.5/site-packages/roslibpy/tf.py", line 95, in update_goal
    request, self._process_response, self._process_error)
  File "/home/anwar/.local/lib/python3.5/site-packages/roslibpy/core.py", line 248, in call
    }), callback, errback)
  File "/home/anwar/.local/lib/python3.5/site-packages/roslibpy/ros.py", line 143, in send_service_request
    self.factory.on_ready(_send_internal)
  File "/home/anwar/.local/lib/python3.5/site-packages/roslibpy/comm/comm_autobahn.py", line 74, in on_ready
    callback(self._proto)
  File "/home/anwar/.local/lib/python3.5/site-packages/roslibpy/ros.py", line 140, in _send_internal
    proto.send_ros_service_request(message, callback, errback)
  File "/home/anwar/.local/lib/python3.5/site-packages/roslibpy/comm/comm.py", line 79, in send_ros_service_request
    json_message = json.dumps(dict(message)).encode('utf8')
  File "/usr/lib/python3.5/json/__init__.py", line 230, in dumps
    return _default_encoder.encode(obj)
  File "/usr/lib/python3.5/json/encoder.py", line 198, in encode
    chunks = self.iterencode(o, _one_shot=True)
  File "/usr/lib/python3.5/json/encoder.py", line 256, in iterencode
    return _iterencode(o, 0)
  File "/usr/lib/python3.5/json/encoder.py", line 179, in default
    raise TypeError(repr(o) + " is not JSON serializable")
builtins.TypeError: dict_keys(['turtle1']) is not JSON serializable
```

## Reproduction Steps
```python
from __future__ import print_function
import roslibpy
from roslibpy import Ros
from roslibpy.tf import TFClient
ros = Ros("localhost", port=9090)
def init_engine():
    #doing unrelated things here 
    print ("started")
    pass
ros.on_ready(init_engine)
tf_client = TFClient(ros, fixed_frame='world')
tf_client.subscribe(frame_id="/turtle1", callback=print)
ros.run_forever()
```

## System Information
running on ubuntu 16.04, python3, and roslibpy version 0.2.0

gonzalocasas, 2018-07-31:
@Behery thanks for reporting, I'll take a look

gonzalocasas, 2018-07-31:
@Behery fixed it and released a patch version: `0.2.1`

Behery, 2018-07-31:
awesome! thanks :) :+1: