RobotWebTools/roslibpy #8: how can i avoid use run_forever() function

loulansuiye, 2018-08-07:
run_forever()[source]

    Kick-starts a blocking loop to wait for events.

    Depending on the implementations, and the client applications, running this might be required or not.

the following sentence says that the user do not have to use this api ,how can reach it?Thank you very much

gonzalocasas, 2018-08-07:
@loulansuiye if you use `roslibpy` from .NET (i.e. IronPython) the implementation is different and in that case, there's less reasons to use it. If you use it from CPython, you need to invoke the start of the event loop in one way or another. It doesn't have to be calling the `run_forever` function thou, but since the event loop is controlled by Twisted reactors, you need to start the reactor somehow.

If your issue is the blocking part, the solution is simply to run this on a thread -which implies you need to control the event loop of your application in some other way. For a detailed example of how to do this, you can check this comment: https://github.com/gramaziokohler/roslibpy/issues/7#issuecomment-410455798

Hope it helps!

gonzalocasas, 2019-01-23:
Closed with #15

closed 2019-01-23 by gonzalocasas