RobotWebTools/roslibpy #9: unsubscribe from topic fails in ghpython

Kathrin3010, 2018-08-21:
Summary.

## Expected Result
subscribing and unsubscribing from a topic with a toggle in ghpython

## Actual Result
the same example that I am running in python (subscribing-unsubscribing-subscribing) cannot be reproduced with ironpython in gh, in which I encountered the following error:
```
Runtime error (KeyNotFoundException): KeyError

Traceback:
  line 160, in pop, "C:\Program Files\Rhino 6\Plug-ins\IronPython\Lib\collections.py"
  line 219, in off, "C:\Users\kathrind\workspace\libraries\roslibpy\src\roslibpy\event_emitter.py"
  line 90, in off, "C:\Users\kathrind\workspace\libraries\roslibpy\src\roslibpy\ros.py"
  line 137, in unsubscribe, "C:\Users\kathrind\workspace\libraries\roslibpy\src\roslibpy\core.py"
  line 63, in script
```

## Reproduction Steps
see files attached, i ran the files in Windows subscribing to a chatter publisher running in in ROS in Ubuntu. subscribing and unsubscribing in gh python is done over the toggle "Subscribe".

[testfiles.zip](https://github.com/gramaziokohler/roslibpy/files/2306923/testfiles.zip)

## System Information
Windows 10, ROS kinetc

gonzalocasas, 2018-08-28:
@Kathrin3010 this is actually caused by the way Grasshopper scripting works, not `roslibpy`. The callback subscription is done with the method `receive_message_chatter`, but that method is created on the same GhPython component where the subscription occurs. That means that every time Grasshopper runs the component, the instance of the method is different (although identical in behavior).

The unsubscribe method needs the same method instance to unsubscribe the callback, so, when the code runs again, and calls `unsubscribe`, the method instance passed is not the same as the subscribed one, causing eventually the error.

We should improve the error message, because `KeyError` is not exactly telling much thou.

One easy fix is to move the callback method to an external python file, because in that case, the instance is only one.

Say you put the following in an importable file called `ros_callbacks.py`:
```python
import scriptcontext as sc

def receive_message_chatter(message):
    sc.sticky['message_chatter'] = message
```

Then in the subscribe/unsubscribe parts of your code, you do:
```python

import ros_callbacks

listener.subscribe(ros_callbacks.receive_message_chatter)

# [..]

listener.unsubscribe(ros_callbacks.receive_message_chatter)
```

And that should do the trick.

gonzalocasas, 2018-08-28:
If you agree, I'll close this and open a new one to improve the error message.

closed 2018-08-30 by gonzalocasas

gonzalocasas, 2018-08-30:
@Kathrin3010 just published a new release, now it's no longer required to pass the callback instance again, since it cannot be a different one than the subscribed callback, so, the method does not need any param.

commit 26d011e3 (2018-08-30): close #9: remove the need to pass the original callback to `listener.unsubscribe()`