RobotWebTools/roslibpy #10: Reconnect method and kill a running while True loop

mrlowalowa, 2018-08-27:
Hey,

I'm struggeling a bit to write a reconnect method if the rosbridge running on my roscore is killed and relaunched.

My problem is that the function from ros.py is_connected() is not working fast enough/relaiable so it thinks it is disconnected while is already said it had a connection again. I had to include a wait period before the next potential reconnect..

```
if not ros_client.is_connected and not reconnecting:
			reconnecting = True
			print ("not connected anymore")
			ros_client.close()
			print ("connection closed")
			print ("Is connected? " + str(ros_client.is_connected))
			print ("Try to reconnect")
			time.sleep(5)
			ros_client.connect()
			while not ros_client.factory.is_connected:
				time.sleep(5)
				print ("Wait for reconnect")
				# ros_client.close()
				# ros_client.connect()
				# print (ros_client.is_connected)
				# print (ros_client.id_counter)
				pass
			print ("Reconnected")
			recon_time = time.time()

		if reconnecting:
			if ( time.time() - recon_time) > 20:
				reconnecting = False
```

This is happening in a while loop. A potential ROS message containing IMUdata is sent right before this.

Another Error: I can not <CTRL> - <C> kill programs running the while loop started by run_forever().

Do I have to manually capture the <CTRL> - <C> Signal and issue a ros_client.terminate() and a exit() afterwards. A normal exit() inside my while True: ... is also not working.

gonzalocasas, 2018-08-28:
@mrlowalowa do you run this on IronPython or in CPython?

gonzalocasas, 2018-08-28:
@mrlowalowa actually, on CPython, `roslibpy` already reconnect automatically. Maybe you need to re-hook your events. Check this sample code:

```
from __future__ import print_function

if __name__ == '__main__':
    import roslibpy
    import logging

    logging.basicConfig(
        level=logging.DEBUG, format='[%(thread)03d] %(asctime)-15s [%(levelname)s] %(message)s')
    LOGGER = logging.getLogger('test')

    ros = roslibpy.Ros(host='localhost', port=9090)
    ros.on_ready(lambda: LOGGER.info('Is ROS connected? %s', ros.is_connected))
    ros.run_forever()
```

Which produces the following output:

```
$ python examples\test_reconnect.py
[22772] 2018-08-28 13:55:23,234 [DEBUG] Started to connect...
[22772] 2018-08-28 13:55:23,254 [DEBUG] Server connected: tcp4:127.0.0.1:9090
[22772] 2018-08-28 13:55:23,254 [INFO] Connection to ROS MASTER ready.
[25956] 2018-08-28 13:55:23,255 [INFO] Is ROS connected? True
[22772] 2018-08-28 13:55:26,873 [INFO] WebSocket connection closed: Code=1006, Reason=connection was closed uncleanly (peer dropped the TCP connection without previous WebSocket closing handshake)
[22772] 2018-08-28 13:55:26,873 [DEBUG] Lost connection. Reason: [Failure instance: Traceback (failure with no frames): <class 'twisted.internet.error.ConnectionDone'>: Connection was closed cleanly.
]
[22772] 2018-08-28 13:55:26,875 [INFO] <twisted.internet.tcp.Connector object at 0x000002B1A0DDADA0> will retry in 2 seconds
[22772] 2018-08-28 13:55:26,875 [INFO] Stopping factory <roslibpy.comm.comm_autobahn.AutobahnRosBridgeClientFactory object at 0x000002B1A0D1D5C0>
[22772] 2018-08-28 13:55:29,128 [INFO] Starting factory <roslibpy.comm.comm_autobahn.AutobahnRosBridgeClientFactory object at 0x000002B1A0D1D5C0>
[22772] 2018-08-28 13:55:29,132 [DEBUG] Started to connect...
[22772] 2018-08-28 13:55:30,138 [DEBUG] Connection failed. Reason: [Failure instance: Traceback (failure with no frames): <class 'twisted.internet.error.ConnectionRefusedError'>: Connection was refused by other side: 10061: No connection could be made because the target machine actively refused it..
]
[22772] 2018-08-28 13:55:30,142 [INFO] <twisted.internet.tcp.Connector object at 0x000002B1A0DDADA0> will retry in 6 seconds
[22772] 2018-08-28 13:55:30,145 [INFO] Stopping factory <roslibpy.comm.comm_autobahn.AutobahnRosBridgeClientFactory object at 0x000002B1A0D1D5C0>
[22772] 2018-08-28 13:55:36,364 [INFO] Starting factory <roslibpy.comm.comm_autobahn.AutobahnRosBridgeClientFactory object at 0x000002B1A0D1D5C0>
[22772] 2018-08-28 13:55:36,365 [DEBUG] Started to connect...
[22772] 2018-08-28 13:55:36,371 [DEBUG] Server connected: tcp4:127.0.0.1:9090
[22772] 2018-08-28 13:55:36,372 [INFO] Connection to ROS MASTER ready.
```

Let me know if that works for you.

gonzalocasas, 2018-09-11:
@mrlowalowa ping ;)

gonzalocasas, 2019-04-10:
@mrlowalowa did you ever had time to look into this? we fixed some reconnection issues on the latest release (`0.4.1`)

gonzalocasas, 2019-05-03:
Since there's no activity, and we fixed several reconnection issues on the latest release, I'm going to assume it's fine to close this. If you still find the issue, please re-open; also, feel free to keep commenting on it even if closed.

closed 2019-05-03 by gonzalocasas