RobotWebTools/roslibpy #11: Publishing and subscribing on custom messages

aakashsehgal, 2018-10-01:
Summary.
I am trying to use roslibpy to publish and subscribe custom messages to/ from a ROS machine.

I try to publish on a custom topic using a custom_message but it doesnt work.

## Expected Result
I had expected that when I use `rostopic list` command on my ROS machine, then it should have showed me `/custom_topic` in the list but unfortunately it doesnt. 

Am I missing something here

## Actual Result
It shows me the default 2 topics i.e. /rosout and /rosout_agg

## Reproduction Steps
```python
This is my sample code : 

`from __future__ import print_function
import time
import roslibpy
from roslibpy import Message, Ros, Topic

def run_topic_pub():
    print('entering run_topic_pub')
    ros = roslibpy.Ros(host='172.17.0.2', port=9090)
    publisher = Topic(ros, '/custom_topic', 'custom_message/float32')

    def start_sending():
        while True:
            if not ros.is_connected:
                break
            publisher.publish(Message({'data': '1'}))
            time.sleep(0.1)
        publisher.unadvertise()

    
    ros.on_ready(lambda: print('Is ROS connected ?', ros.is_connected))
    ros.on_ready(start_sending, run_in_thread=True)
    ros.get_topics(print)

    ros.run_forever()

if __name__ == '__main__':

    print('main')

    run_topic_pub()
`

```

## System Information
Linux (ubuntu 16.04), ROS Melodic

aakashsehgal, 2018-10-01:
I looked at the documentation more carefully and found that we need to define the topic first. So i did the following before creating a publisher : 

`message = roslibpy.Message()
topic = roslibpy.Topic(ros,custom_topic,message)
publisher = Topic(ros,/custom_topic, std_msgs/ String)`

and it works.

However I am still not able to add a custom message something like : 

`message = roslibpy.Message('custom_message/float32')
topic = roslibpy.Topic(ros,/custom_topic,message)
publisher = Topic(ros,'/custom_topic,'custom_message/float32'')`

How can i basically create a custom message definition ?

gonzalocasas, 2018-10-02:
@aakashsehgal If what you want is really just a float32, you can use `std_msgs/float32` (http://docs.ros.org/lunar/api/std_msgs/html/msg/Float32.html) instead of a custom one, but I assume that was just used as example. If you need a custom message type, you need to create it and compile as usual in ROS. As far as I know, there's no way for ROS to dynamically build a message type.

aakashsehgal, 2018-10-02:
@gonzalocasas yes true that was just an example.
I have already created and compiled ROS with this custom message. So basically it is available in ROS. But since I am using the rosbridge to connect to a non ROS machine, how can i use the same custom message definition in roslibpy ?

gonzalocasas, 2018-10-02:
@aakashsehgal that should work, but you need to stop `rosbridge` re-run the `source` of your catkin workspace and restart `rosbridge`. After that, the new message type should just work.

aakashsehgal, 2018-10-02:
@gonzalocasas thank you for the information. I will try this out day after tomorrow and let you know.
E.g if my message type is custom_msg/ float 32 then would it work In the similar way as mentioned above ?

gonzalocasas, 2018-10-02:
@aakashsehgal yep, it should work fine.

aakashsehgal, 2018-10-08:
@gonzalocasas : Somehow it doesnt work. The rosbridge gives an error : `Unable to load the manifest for package <pkg_name>`
Could you please refer [here](https://answers.ros.org/question/304877/rosbridge-unable-to-load-the-manifest-for-the-package/)

aakashsehgal, 2018-10-08:
I have the following message structure : 

Msg1.msg
    int a
    int b
    float c
    float d
    float e

Msg2.msg
   int x
   Msg1[] y

I have 2 ros nodes, where in the first one publishes and the second one subscribes : both on topic `Msg2`.
Just for a test, now I am publishing the message using the 1st node in ROS and subscribing using ROS_BRIDGE. How can I use it in the ROSBridge_Python_API : I am doing something like this : 

topic = Topic(ros, /custom_topic, 'Msg2[a,b,c,d,e], Msg2/int')

But i guess I am doing something wrong here.

gonzalocasas, 2018-10-08:
I'm going to close this because I think it's not a bug in `roslibpy`, but feel free to keep discussing it here anyway. And if you think it is actually a bug in `roslibpy`, then re-open pls.

closed 2018-10-08 by gonzalocasas

deep-saxena, 2018-10-08:
I am also trying to publish to a ROS topic using roslibpy. I have used the similar code for publishing as given in the summary above and have also added the logger using:
 
    import logging

    logging.basicConfig(level=logging.DEBUG, format='[%(thread)03d] %(asctime)-15s [%(levelname)s] %(message)s')
    LOGGER = logging.getLogger('test')

I am facing issue as the `/custom_topic` is not showing when I run the  `rostopic list` command after running this script. ros_bridge node is already running at port 9090 using the command `roslaunch rosbridge_server rosbridge_websocket.launch`.

Looking at the logs it seems that the connection gets closed as soon as we send data over the websocket.

> [140283134949120] 2018-10-08 23:27:29,112 [DEBUG] Started to connect...
> [140283134949120] 2018-10-08 23:27:29,116 [DEBUG] Server connected: tcp4:127.0.0.1:9090
> [140283134949120] 2018-10-08 23:27:29,116 [INFO] Connection to ROS MASTER ready.
> Is ROS connected ? True
> [140283046196992] 2018-10-08 23:27:29,119 [DEBUG] Sending ROS message|<pre>{"topic": "/custom_topic", "queue_size": 100, "latch": false, "type": "std_msgs/String", "id": "advertise:/custom_topic:1", "op": "advertise"}</pre>
> [140283046196992] 2018-10-08 23:27:29,119 [DEBUG] Sending ROS message|<pre>{"topic": "/custom_topic", "msg": {"data": "Hello"}, "latch": false, "id": "publish:/custom_topic:2", "op": "publish"}</pre>
> [140283134949120] 2018-10-08 23:27:29,120 [INFO] WebSocket connection closed: Code=1009, Reason=message too big
> [140283134949120] 2018-10-08 23:27:29,120 [DEBUG] Lost connection. Reason: [Failure instance: Traceback (failure with no frames): <class 'twisted.internet.error.ConnectionLost'>: Connection to the other side was lost in a non-clean fashion.
> ]
> [140283134949120] 2018-10-08 23:27:29,120 [INFO] <twisted.internet.tcp.Connector instance at 0x7f9631a73cb0> will retry in 2 seconds
> [140283134949120] 2018-10-08 23:27:29,120 [INFO] Stopping factory <roslibpy.comm.comm_autobahn.AutobahnRosBridgeClientFactory object at 0x7f9631cf4cd0>
> [140283134949120] 2018-10-08 23:27:31,432 [INFO] Starting factory <roslibpy.comm.comm_autobahn.AutobahnRosBridgeClientFactory object at 0x7f9631cf4cd0>
> [140283134949120] 2018-10-08 23:27:31,433 [DEBUG] Started to connect...
> [140283134949120] 2018-10-08 23:27:31,442 [DEBUG] Server connected: tcp4:127.0.0.1:9090
> [140283134949120] 2018-10-08 23:27:31,442 [INFO] Connection to ROS MASTER ready.

The websocket close code 1009 specifies "endpoint won't process large frames". Can you help me with what could be the possible reason for this.

aakashsehgal, 2018-10-08:
@deep-saxena After compiling your package, source the environment and then launch the rosbridge in the same terminal. This should help.

aakashsehgal, 2018-10-09:
@gonzalocasas : I solved the problem. I was basically trying to find the wrong package. And yes, you mentioned it correctly, I had to use `publisher = Topic(ros, '/custom_topic', 'ros_tutorials/Msg2')`. Now I can subscribe easily.
Could you here please guide me as to how can I publish a message using the same API.
I you look at my message definition above, iin Msg2.msg I have an array of Msg1. How can I do it in an efficient way ?
`publisher.publish(Message)({'x' : '10'}, 'Msg1.a':'20'))`. Something like this wherein I can loop through all the members of the array.

gonzalocasas, 2018-10-09:
@aakashsehgal:

> I solved the problem

cool!

> how can I publish a message using the same API

It's relatively straight-forward, you don't need to add any additional type information, simply supply a message (which is basically a python dictionary) that matches to the expected format.

To exemplify it with a simple message type, if you want to publish a message of type `std_msgs/String` ([see here](http://docs.ros.org/api/std_msgs/html/msg/String.html)), you would publish it as follows:

```python
publisher.publish(Message({'data': 'hello world'}))
```

For your case of `ros_tutorials/Msg2`, I think the message should be:
```python
publisher.publish(Message({'x': 123, 'y': [{'a': 1, 'b': 2, 'c': 3.14, 'e': 7.23, 'f': 2.353}]}))
```
(notice `y` is a list, so you can pass as many `Msg1`-matching dictionaries as you want).

aakashsehgal, 2018-10-09:
@gonzalocasas : that was quite simple :p thanks :)

deep-saxena, 2018-10-09:
@aakashsehgal I was sourcing my catkin_workspace's  setup.bash which seems to have some fault. When I sourced the file from /ope/ros/kinetic/setup.bash I was able to receive the messages on rosbridge server. Thanks.

aakashsehgal, 2018-10-18:
@gonzalocasas : can I also somehow obtain the subscribed messages ? Something like this : 

I am publishing in ROS on topic custom_topic with Msg2 and I need to subscribe these using roslibpy.

```
listener = Topic(ros, 'custom_topic', 'ros_tutorials/Msg2')

def receive_message(message)
    a = message['x']
    print(a)

listener.subscribe(receive_message) 
```

I need to process the messages received. So how can I access these messages ?

aakashsehgal, 2018-10-19:
@gonzalocasas : how can I publish and subscribe simultaneously on 2 different topics. I have two nodes A and B. A is located in ROS and i am using roslibpy to communicate with B.
A publishes first on custom_topic1, B subscribes this, processes it and publishes on `custom_topic2` which is then subscribed by A. A processess this again and publishes back on `custom_topic1`. and then it works in the loop.

I am somehow getting stuck in an infinite loop.

I would want to do something like this : 

```
ros = roslibpy.Ros(host='...', port=9090)
publisher = Topic(ros, '/custom_topic1', 'ros_tutorials/Msg2')
subscriber = Topic(ros, '/custom_topic2', 'ros_tutorials/Msg1')

def publish:
    publisher.publish(Message({'x': 123, 'y': [{'a': 1, 'b': 2, 'c': 3.14, 'e': 7.23, 'f': 2.353}]}))
    subscriber.subscribe(receive_message)
def receive_message(message):
    a = message['j']
    print('a')
```

but i always get stuck in the `receive_message()` function loop and it never comes out. Do you have a better idea ?

gonzalocasas, 2018-10-23:
@aakashsehgal I think you flow is an infinite loop by definition. If I understand correctly:
`A` -> `custom_topic1` -> `B` -> `custom_topic2` -> `A` -> `custom_topic1` (so, A will enter in a loop because it's subscribed to `custom_topic2` but publishes that again to `custom_topic1`.

It really sounds like the problem is not related to code, is has a circular dependency in your node setup.

aakashsehgal, 2018-10-23:
@gonzalocasas : yes, this is my usecase. and indeed it is a circular dependency where it enters the infinite loop. Any leads on how to avoid this ?

ipa-mdl, 2018-10-23:
> Any leads on how to avoid this ?

Many nodes form these two-way communication patterns (incl. actionlib), but it you should avoid unconditional responses, which would result in an infinite loop.
Or you could use a service.

tomasruizt, 2018-11-16:
@aakashsehgal One thing I noticed about your script, instead of writing:
```python
while True:
    if not ros.is_connected:
        break
```
use
```python
while ros.is_connected:
    # do something
```