RobotWebTools/roslibpy #14: How can I publish the messages at a certain frequency ?

aakashsehgal, 2019-01-14:
I have a subscriber node in ROS where a callback is implemented which makes it dependent on the frequency of the publisher. Now I want to publish messages to this subscriber node through rosbridge and i am using roslibpy for that. 

Currently I am using the sleep function but i want to avoid using that.

What is basically the `throttle_rate` parameter. I tried to play with this but did not help. Is there some `ros::rate()` type of functionality with which i can send messages at a certain fixed frequency ?

Also, if no sleep is used, then what is the default rate of publishing ?

When I use `rostopic hz` on the ros side, then it shows me somewhere around `850ms`

## System Information
Windows 10, ROS melodic

rarmstrong22, 2019-01-28:
+1 on the `throttle_rate ` param appearing to have no effect. See https://answers.ros.org/question/313844/roslibpy-why-does-tfclient-seem-to-ignore-the-rate-parameter/

ipa-mdl, 2019-01-28:
> What is basically the throttle_rate parameter

`throttle_rate` is only used for [subscriptions](https://github.com/RobotWebTools/rosbridge_suite/blob/develop/ROSBRIDGE_PROTOCOL.md#344-subscribe).

> Is there some ros::rate() type of functionality with which i can send messages at a certain fixed frequency ?

One option: [Timer](https://docs.python.org/2/library/threading.html#timer-objects).

rarmstrong22, 2019-01-28:
> Beware, it is not actually a rate, but the interval in milliseconds!

Ah! Is this also true for the `rate` paramter to `TFClient`?

ipa-mdl, 2019-01-28:
> Ah! Is this also true for the rate paramter to TFClient

No, I am talking about `throttle_rate`. Your problem is unrelated to this issue.

rarmstrong22, 2019-01-28:
> No, I am talking about throttle_rate. Your problem is unrelated to this issue.
Just looked in `tf2_web_republisher` and I can see it now; thanks.

nageshpindi, 2019-01-29:
Hi,
     I'm getting error at **ros.run()**below is the error.

Traceback (most recent call last):
  File "ros-service-caller.py", line 4, in <module>
    client.rosrun()
AttributeError: 'Ros' object has no attribute 'rosrun'