RobotWebTools/roslibpy #16: Subscribe CompressedImage has 10~20 seconds delay

leepengcheng, 2019-01-27:
Summary.

## Expected Result
I am trying to use roslibpy to subscribe the CompressedImage topic pubish by ros

## Actual Result
the image has almost 10~20 seconds delay

but when use rviz,the image works well

## Reproduction Steps
```python
import sys
sys.path.remove('/opt/ros/kinetic/lib/python2.7/dist-packages')
import roslibpy
import numpy as np
import cv2
import base64
import time

client = roslibpy.Ros(host='localhost', port=9090)
listener = roslibpy.Topic(client, '/simulator/sensor/camera/left/compressed', 'sensor_msgs/CompressedImage',queue_size=1)
cv2.namedWindow("ros")

def start_listening():
    listener.subscribe(receive_message)

def receive_message(message):
    start=time.time()
    data=base64.b64decode(message["data"])
    np_img = np.frombuffer(data, np.uint8)
    img=cv2.imdecode(np_img,cv2.IMREAD_COLOR)
    end=time.time()
    print("fps is %.f"%((1.0/(end-start))))
    cv2.imshow("ros",img)
    cv2.waitKey(5)

client.on_ready(start_listening, run_in_thread=True)
client.run_forever()
```

## System Information
Linux (ubuntu 16.04), ROS Kinetic

ipa-mdl, 2019-01-27:
In general `rosbridge` does not perform very well with image data, because the binary data has to be converted into the JSON format (and back).
That's one of the reasons for using [web_video_server](http://wiki.ros.org/web_video_server) .

However, 10-20 seconds delay is quite big..
How did you measure these numbers?
The provided code measures the time for decoding and the button press, but not the delay.
You should move the actual work somewhere else to not block the handler..

leepengcheng, 2019-01-28:
> In general `rosbridge` does not perform very well with image data, because the binary data has to be converted into the JSON format (and back).
> That's one of the reasons for using [...]

I drive car in [LGSVL simulator](https://github.com/lgsvl/simulator),and pubish the virtual compressedImage data with ros，this process lasted for nearly 20 seconds。
When the target position is reached in the simulator，the roslipy subscribed image   just started。

ipa-mdl, 2019-01-28:
TL;DR: If possible, just use `rospy`. If not, throttle the image subscription.
https://github.com/lgsvl/simulator/issues/70 might be related.

> When the target position is reached in the simulator，the roslipy subscribed image just started。

I cannot reproduce this. I see a build-up of delay, probably depending on image size and frequency.

leepengcheng, 2019-01-29:
> TL;DR: If possible, just use `rospy`. If not, throttle the image subscription.
> [lgsvl/simulator#70](https://github.com/lgsvl/simulator/issues/70) might be related.
> 
> > When the target position is reached in the simulator，the roslipy subscribed image just started。
> 
> I cannot reproduce this. I see a build-up of delay, probably depending on image size and frequency.

i set `throttle_rate=50`,now the delay is about 2s. I think it's the problem of simulator.

closed 2019-01-29 by leepengcheng