RobotWebTools/roslibpy #17: I'm getting Error

nageshpindi, 2019-01-29:
Hello guys,
              I'm new to roslibpy and i'm getting below error can any one give me some advices.
>>> ros.run()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'Ros' object has no attribute 'run'
Thanks in advance.

Summary.

## Expected Result
What you expected.

## Actual Result
What happened instead.

## Reproduction Steps
```python
import roslibpy

```

## System Information
Operating System name and version, ROS version, etc.

gonzalocasas, 2019-01-29:
It looks like you have an outdated version of `roslibpy`. The `run()` method was added on version `0.4.0`.
Please update it and try again.
I will close this because it's not a code bug, but feel free to keep commenting here if it doesn't work for you.

closed 2019-01-29 by gonzalocasas

nageshpindi, 2019-02-27:
Hi Casas,
I have some doubts in roslibpy. So how to call more than one service in a
single python file. The success_callback function being called but its not
exiting the success_callback function.
below is the code
--------------------------

import roslibpy
import logging

client = roslibpy.Ros(host='localhost', port=9090)
client.run()

client.on_ready(lambda: call_service())

service = roslibpy.Service(client,
'/rosplan_problem_interface/problem_generation_server', 'std_srvs/Empty')

logging.basicConfig( level=logging.DEBUG, format='[%(thread)03d]
%(asctime)-15s [%(levelname)s] %(message)s')
LOGGER = logging.getLogger('test')

def call_service():
    print('Calling service')
    request = roslibpy.ServiceRequest()
    service.call(request, success_callback, error_callback)
    return
def success_callback(result):
    print('Service response: ')
    return
def error_callback(result):
    print('Something went wrong')
    return
try:
    while True:
        pass
except KeyboardInterrupt:
    pass

client.terminate()

according to that code can u explain how to call more than one service.
thanks in advance
Nagesh

nageshpindi, 2019-02-27:
I have one more doubt i'm importing roslibpy package but one of its method
is not importing called "from . comm import RosBridgeClientFactory" i'm
getting error i have tried many ways still i'm able to solve that error.
Error
--------------------------------------------------
from . comm import RosBridgeClientFactory
ValueError: Attempted relative import in non-package

Thanks and regards
Nagesh.