RobotWebTools/roslibpy #19: Use this libary with the "ros2-web-bridge" project

LeptonByte, 2019-02-21:
Is it possible to use this libary with the [ros2 webbridge](https://github.com/RobotWebTools/ros2-web-bridge)? 

I want to connect my EV3 robot with ros2 nodes on my pc. _I wasn't able to run/build ros2 on the EV3_, so I searched for an alternative and found **this awsome project** in addition to the **ros2-web-bridge**.
So what do you think? Is it possible?

best regards

## System Information
Windows 10, ROS 2 Bouncy Bolson
and
ev3dev - Debian Jessie

gonzalocasas, 2019-02-21:
In principle, yes, the ros bridge protocol hasn't changed so it should work.

gonzalocasas, 2019-02-21:
I will close this issue since it is not actionable right now, and in principle, it should "just work", but if you find any issues when trying it out, please do open a new one, or comment on this one.

closed 2019-02-21 by gonzalocasas