RobotWebTools/roslibpy #22: sending custom messages, error occurs, but still work

XinArkh, 2019-04-12:
I was learning to send messages other than the `std_msgs/String` type shown in the doc. Inspired by [your reply](https://github.com/gramaziokohler/roslibpy/issues/11#issuecomment-428178646) in #11, I tried sending `std_msgs/Float64MulitArray` type messages by rewriting the Message dict parameter. 

## Actual Result
The system works anyhow. However, there's an error occurs in the master console and I have no idea why and whether there are any unexpected influences of it.

[image]

## Reproduction Steps
```python
import roslibpy
import time

client = roslibpy.Ros(host='10.13.106.70', port=9090)
talker = roslibpy.Topic(client, '/chatter', 'std_msgs/Float64MulitArray')
# talker = roslibpy.Topic(client, '/chatter', 'std_msgs/String')

def start_talking():
    t = time.time()
    while client.is_connected:
        # talker.publish(roslibpy.Message({'data': 'Hello World! {}'.format(time.time()-t)}))
        talker.publish(roslibpy.Message({'layout': {'dim': [{'label': 'test', 'size': 3, 'stride': 1}], 
                                                    'data_offset': 0}, 
                                         'data': [1.0, 2.0, 3.0]}))
        print('Sending message...%d' %(time.time()-t))
        time.sleep(1)

    talker.unadvertise()

client.on_ready(start_talking)
client.run_forever()

```

## System Information
talker: win10, python3.6
ros master: ubuntu16

## Additional Question
There is one more detailed confusion I could hardly understand. 
In ros wiki, `tf2_web_republisher` is functioned to precompute tf transform information. I thought I didn't need tf transform, so I started ros master without `tf2_web_republisher`. As a result, the connection was failed. I wonder why `tf2_web_republisher` has something to do with a connection that doesn't need tf transform function. 

Thanks a lot for your reply!

ipa-mdl, 2019-04-12:
It does not work, rosbridge cannot process the data, because it the type is unknown.

It looks like you have a typo in the message type: std_msgs/Float64Mul**it**Array.

XinArkh, 2019-04-12:
@ipa-mdl Yes, it's my fault!🤣Feeling like a fool...
So can you explain the confusion in the additional question? Thank you very much!

gonzalocasas, 2019-04-12:
I'll close this one since it's not an actual issue, but feel free to keep commenting on it.

closed 2019-04-12 by gonzalocasas