RobotWebTools/roslibpy #23: [feature request] Run roslibpy on google colab

talregev, 2019-04-12:
I was trying to get run roslibpy on google colab, and it not connect.
I run roscore in the background.
I run hello world ros python node in the back ground of string.
I can see the topic with rostopic list, and see the data in rostopic echo,
but `ros.is_connected` is `False`
Can you help?
[roslibpy on google colab][1]

[1]:https://colab.research.google.com/drive/1oUjv4MDwKRmX94097zTvszm_0pYI5rLX

gonzalocasas, 2019-04-12:
Calling `ros.is_connected` immediately after `ros.run()` will usually return false because `run` is a non-blocking call so when you check for connection, it's not connected **yet**. 

This behavior is documented here: https://roslibpy.readthedocs.io/en/latest/examples.html#waiting-for-connection 
Try using the `on_ready` callback to detect connection and work with it. I haven't really tried roslibpy in a jupyter notebook, but perhaps, if you check the connection status from another cell, that would also work?

gonzalocasas, 2019-04-22:
@talregev any news? did it work?

talregev, 2019-04-25:
It didn't work.     
I didn't see the messages on colab.      
Not sure how to solve it, I share the link in the first message that people can play it and share if succeeded.

gonzalocasas, 2019-05-29:
@talregev I just checked it again and noticed you were not installing `rosbridge` on the first cell. Here's an [updated notebook](https://colab.research.google.com/drive/1kbQcEw1eYHfiKdQjChToEdAy5jacxVwU) where I fixed that plus a little fix on your last cell as well.

[image]

So, I'm going to close this issue. But feel free to continue commenting on it if you have questions.

closed 2019-05-29 by gonzalocasas

talregev, 2019-06-29:
Thank you! it was working! :)    
But still there is problem:    
I take your code from the example, and I create example of talker and listener in google colab.     
You can run this example one time. when you try to run it again, it throw an exception.     
Can you help me what need to close / kill process that the example will run again? 
It also happen in your example you give me.   

In Other topic,
I think it good for your project to show the roslibpy is working on google colab, because it will be more
easy for people to run it online without any environment installation on their computer for free.
They can see your example for using the lib, they can play with it online, or even start develop with the lib.

So I start this first example, and if you like it, and if it will work more then once, I can commit it to the docs under google colab example lib and transfer all the example to working example on google colab.

What do you say? 

```bash
Exception in thread Thread-6:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/lib/python2.7/dist-packages/twisted/internet/base.py", line 1242, in run
    self.startRunning(installSignalHandlers=installSignalHandlers)
  File "/usr/lib/python2.7/dist-packages/twisted/internet/base.py", line 1222, in startRunning
    ReactorBase.startRunning(self)
  File "/usr/lib/python2.7/dist-packages/twisted/internet/base.py", line 730, in startRunning
    raise error.ReactorNotRestartable()
ReactorNotRestartable

ExceptionTraceback (most recent call last)
<ipython-input-10-c353331fa439> in <module>()
      4 
      5 client = roslibpy.Ros(host='localhost', port=9090)
----> 6 client.run(5)
      7 print('Connected: {}'.format(client.is_connected))
      8 

/usr/local/lib/python2.7/dist-packages/roslibpy/ros.pyc in run(self, timeout)
     78 
     79         if not wait_connect.wait(timeout):
---> 80             raise Exception('Failed to connect to ROS')
     81 
     82     def run_forever(self):

Exception: Failed to connect to ROS
```

gonzalocasas, 2019-07-04:
> So I start this first example, and if you like it, and if it will work more then once, I can commit it to the docs under google colab example lib and transfer all the example to working example on google colab.

Sure, that would be greatly appreciated!

talregev, 2019-07-04:
Can you help me with the error? Should I open a separate issue for that?

gonzalocasas, 2019-07-05:
@talregev the error message is actually very specific, it says `ReactorNotRestartable` and that's exactly the problem. Twisted reactors are not restartable. A reactor is started once during the life-time of an application, and it seems Google Colab tries to reuse the same process or something like that, so, if you do `terminate` and then try to `run` again, that will not work. In order to avoid the problem, do not use `client.terminate()`, instead use `client.close()` which will disconnect but leave the reactor running if needed again.

talregev, 2019-07-05:
Thank you for time.    
I try `client.close()`    

This is the error when I run the code in [google colab][1] second time:
```
ExceptionTraceback (most recent call last)
<ipython-input-10-7898514f213b> in <module>()
      4 
      5 client = roslibpy.Ros(host='localhost', port=9090)
----> 6 client.run(5)
      7 print('Connected: {}'.format(client.is_connected))
      8 

/usr/local/lib/python2.7/dist-packages/roslibpy/ros.pyc in run(self, timeout)
     78 
     79         if not wait_connect.wait(timeout):
---> 80             raise Exception('Failed to connect to ROS')
     81 
     82     def run_forever(self):

Exception: Failed to connect to ROS
```

[1]:https://colab.research.google.com/drive/1oUjv4MDwKRmX94097zTvszm_0pYI5rLX