RobotWebTools/roslibpy #25: External action server possible?

HeinzBenjamin, 2019-04-29:
Hello there and thanks for the great work on roslibpy,

upon reviewing your documentation, I noticed that it's possible to call an action server running inside ROS from outside with the class roslibpy.actionlib.ActionClient. Is it also possible to set up an action server that is running outside ROS with roslibpy that serves calls coming from other ros nodes?

Again thanks for the great work
Best
Benjamin

gonzalocasas, 2019-05-03:
Yes, this is definitely possible. Action servers are basically just a collection of 5 coordinated topics, so it can easily be implemented. Ideally, we could implement a `SimpleActionServer` class to make it easier to do so, but it is also doable without it (For reference, a `SimpleActionServer` using roslibjs [is available here](https://github.com/RobotWebTools/roslibjs/blob/develop/src/actionlib/SimpleActionServer.js), the python implementation would be basically the same).

gonzalocasas, 2019-05-31:
@HeinzBenjamin check out the new release (`0.6.0`), we added an implementation of the simple action server, plus some other improvements (see [here some examples](https://roslibpy.readthedocs.io/en/latest/examples.html#actions) and [check the changelog](https://roslibpy.readthedocs.io/en/latest/changelog.html)).

I will close this, but feel free to keep commenting.

closed 2019-05-31 by gonzalocasas

HeinzBenjamin, 2019-07-17:
Awesome thanks!