RobotWebTools/roslibpy #26: Passing arguments to service

judiet, 2019-05-09:
Hello guys thank for your great work with roslibpy,

I'm trying to send data from a python script to a ROS service. I have trouble to pass the data to the service. Is there an opportunity  to do so? 
```python
import roslibpy

client = roslibpy.Ros(host='localhost', port=9090)
client.run()
client.on_ready(lambda: call_service())

service = roslibpy.Service(client, '/distance_data', 'robocrawler/distance_data')

def call_service():
    #Trying to pass data here
    request = roslibpy.ServiceRequest()
    service.call(request, success_callback, error_callback)
  
def success_callback(result):
    print('Service response: ', result['dist'])

def error_callback(result):
    print('Something went wrong')

try:
    while True:
        pass
except KeyboardInterrupt:
    pass

client.terminate()

```

Thank you in advance.

gonzalocasas, 2019-05-09:
Yes, passing values to the service is easy, just need to pass a dictionary to the service request constructor, i.e. `request = roslibpy.ServiceRequest(dict(val1=3.14, some_other_val=dict(foo="bar")))`

The dictionary you pass in to it needs to match the fields and data types expected by the ROS service you are calling.

judiet, 2019-05-09:
Thank you very much for your quick response. 
You saved us the day. 
I close the issue.

closed 2019-05-09 by judiet

PINKY0212, 2020-11-19:
helloo guyss,
I'm trying to send argument  from a python script to a ROS service. I have trouble to pass the data to the service. Is there an opportunity to do so?
#! /usr/bin/env python
import rospy
from pkg_vb_sim.srv import vacuumGripper
from pkg_vb_sim.srv import vacuumGripperRequest
from pkg_vb_sim.srv import vacuumGripperResponse

from pkg_vb_sim.msg import LogicalCameraImage
        rospy.init_node('node_client')
        sos=rospy.ServiceProxy('/eyrc/vb/ur5_1/activate_vacuum_gripper', vacuumGripper)
        sos.wait_for_service()
        request = vacuumGripper(activate_vacuum_gripper = "true")
        sos.call(request)
please help..where am i going wrong??
thanks in advance

gonzalocasas, 2020-11-19:
@PINKY0212  I think you have the wrong repository. You are using `rospy` (the official python client for ROS within a ROS network), but this repository is for `roslibpy` (a library python library that connects to ROS via websockets from outside a ROS network).