RobotWebTools/roslibpy #29: Ros client does not automatically reconnect to Ros Master

Torgenn, 2019-06-12:
If the rosmaster and the rosbridge node are killed and then restarted, from what I understand from #10, the ros client should automatically reconnect. However, in the example I have implemented it does not reconnect automatically, and I can't figure out a method to make it reconnect.   
I am trying to implement a GUI for a robot running ROS and everything is working great, except every time I stop the rosmaster and the rosbridge node to make some changes I also have to restart the GUI since I couldn't figure out how to make it reconnect.

## Reproduction Steps
I made a small example that demonstrates my use case
```python
import roslibpy
import logging
import sys
from PySide2.QtCore import QObject, Signal, Slot
from PySide2.QtWidgets import QApplication, QWidget
from PySide2.QtGui import QColor

class myRosClient(QObject):
    new_status = Signal(bool)
    def __init__(self,parent=None):
        super(myRosClient, self).__init__(parent)
        self.client=None

    def setClient(self):
        self.client = roslibpy.Ros(host='localhost',port=8080)
        self.killed = None

        position_subscriber = roslibpy.Topic(self.client, '/killed', 'std_msgs/Bool')
        position_subscriber.subscribe(self.update_status_killed)

        logging.basicConfig(
            level=logging.DEBUG, format='[%(thread)03d] %(asctime)-15s [%(levelname)s] %(message)s')
        LOGGER = logging.getLogger('test')
        self.client.on_ready(lambda: LOGGER.info('Is ROS connected? %s', self.client.is_connected))

        self.client.run()

    def update_status_killed(self,message):
        if message["data"]!=self.killed:
            self.killed=message["data"]
            self.new_status.emit(self.killed)

class Demo(QWidget):
    def __init__(self, parent=None):
        super(Demo,self).__init__(parent)
        self.setAutoFillBackground(True)
        pal = self.palette()
        pal.setColor(self.backgroundRole(), QColor('black'))
        self.setPalette(pal)

    @Slot()
    def setColor(self, killed):
        if killed:
            pal = self.palette()
            pal.setColor(self.backgroundRole(), QColor('red'))
            self.setPalette(pal)
        else:
            pal = self.palette()
            pal.setColor(self.backgroundRole(), QColor('green'))
            self.setPalette(pal)

if __name__ == '__main__':
    # Create the Qt Application
    app = QApplication(sys.argv)
    # Create and show the form
    window= Demo()
    ros_client=myRosClient()
    ros_client.setClient()
    ros_client.new_status.connect(window.setColor)
    window.setGeometry(500,300,800,600)
    window.show()
    # Run the main Qt loop
    sys.exit(app.exec_())

```

And the output information is:
```
[140073808373504] 2019-06-12 11:36:39,491 [DEBUG] Server connected: tcp4:127.0.0.1:8080
[140073808373504] 2019-06-12 11:36:39,491 [INFO] Connection to ROS MASTER ready.
[140073808373504] 2019-06-12 11:36:39,491 [DEBUG] Sending ROS message|<pre>b'{"throttle_rate": 0, "op": "subscribe", "topic": "/killed", "id": "subscribe:/killed:1", "type": "std_msgs/Bool", "compression": "none", "queue_length": 0}'</pre>
[140073799980800] 2019-06-12 11:36:39,492 [INFO] Is ROS connected? True
[140073808373504] 2019-06-12 11:37:15,801 [INFO] WebSocket connection closed: Code=1006, Reason=connection was closed uncleanly (peer dropped the TCP connection without previous WebSocket closing handshake)
[140073808373504] 2019-06-12 11:37:15,801 [DEBUG] Lost connection. Reason: [Failure instance: Traceback (failure with no frames): <class 'twisted.internet.error.ConnectionDone'>: Connection was closed cleanly.
]
[140073808373504] 2019-06-12 11:37:15,802 [INFO] Stopping factory <roslibpy.comm.comm_autobahn.AutobahnRosBridgeClientFactory object at 0x7f658ec1ac18>
```  

When I restart the rosmaster and the rosbridge node nothing changes. I tried to terminate he connection in the GUI and reconnect but I was running in a Twisted Reactor error. Is it possible to create a new connection?

## System Information
Ubuntu 16.04, roslibpy 0.5.0, Ros Kinetic, CPython 3.5.2

gonzalocasas, 2019-06-14:
@Torgenn thanks for the report and the repro snippet! I'll look into it

gonzalocasas, 2019-06-19:
Quick update: the main reason why it does not reconnect is that the connection is actually closing cleanly (see the last two messages in your log output) but in retrospect, regardless of the disconnection reason, it would probably make sense to keep retrying. There's a second issue thou, even if the reconnect would work (which is basically just commenting out two lines that explicitly check for clean disconnects), the topic would not re-subscribe automatically.

To match the current implementation on `roslibjs`, I will add the capability to reconnect to topics and action clients and so on, and will see how to handle the disconnect reason, so that even a clean disconnect could end up in a reconnect attempt.

Finally, the Twisted reactor error you mention I guess is that you tried to terminate the reactor and then run it again, which Twisted doesn't really like. But a connection can be closed and re-opened without having to restart the Twisted reactor.

Torgenn, 2019-06-27:
Thanks! To open and close the connection do you use the close() and connect function() of the Ros class?

Torgenn, 2019-06-27:
```
self.client.close()
self.client.connect()
```
Triggered by the user, followed by reconnecting to all the topics did the trick. Thanks!

HeinzBenjamin, 2019-09-13:
Hi I wondered if there's an update to the issue? I'm facing a similar problem that my Service Requests are not answering anymore after I lost connection to the ros core. The following code loops for ever as soon as ros core goes out of service and is restarted.

```
self.srv_proxy = roslibpy.Service(self.client, '/srv', '/my_srv')
request = roslibpy.ServiceRequest(data)
result = None
while result == None:
   try:
      result = self.srv_proxy.call(request)
   except:
      print("ROS is gone. Trying to reconnect...")
      time.sleep(5)
      self.client.close() <--optional
      time.sleep(5)
      self.client.connect() <--optional
      result = None
```

I tried to redefine self.srv_proxy in the except part, but to no success.
Best

closed 2020-03-25 by gonzalocasas

gonzalocasas, 2020-03-25:
This has been fixed and released in 1.0

commit ec25772c (2020-03-25): fix #29 & #44: correctly reconnect if the disconnect was not initiated by client