RobotWebTools/roslibpy #32: Help needed in understanding the API documentation

shikharvaish28, 2019-07-17:
I have a ready robot which is currently running on ROS Kinetic. Also, it relies heavily on tf2 and uses ```rospy``` as well as ```roscpp```. I am planning to use ```roslibpy``` to port the python nodes to ```ros2``` but I have stuck some issues. It would be great if anyone of you could help me with those.

```python
    rospy.init_node('ros_mpc')
    rospy.Subscriber("commands/stop", Bool, estop_callback)
    rospy.Subscriber("drive_act", Point32, act_callback)
    pub      = rospy.Publisher('drive_parameters', Point32, queue_size=1)
    pub_path = rospy.Publisher('path', Path, queue_size=1)
    pub_fren = rospy.Publisher('fren_pos', Path, queue_size=1)
    pub_targ = rospy.Publisher('target', Path, queue_size=1)
    thread.start_new_thread(spin, ())
```

To start with, I am not able to find the alternative for ```init_node``` so that I can initiate a node and then the sequence of porting would continue to go on.

gonzalocasas, 2019-07-17:
@shikharvaish28 you don't initialize nodes when using `roslibpy` (or the ros web bridge actually), the node is the bridge itself, and it will provide and consume services, topics, actions through its registration.

Please check the examples in the doc to see how a `roslibpy` node differs from a `rospy`, and also check the `tests` folder for a few more examples.

I'm closing this now, since it's not actionable, but feel free to keep commenting.

closed 2019-07-17 by gonzalocasas

shikharvaish28, 2019-07-17:
Thanks for the prompt reply @gonzalocasas 

```python
def build_path_msg(xs, ys):
    p_msg = Path()
    p_msg.header.frame_id = "map"
    
    for i in xrange(len(xs)):
        p = PoseStamped()
        p.pose.position.x = xs[i] 
        p.pose.position.y = ys[i]
        p_msg.poses.append(p)
    return p_msg

pub_fren = roslibpy.Topic(client, 'fren_pos', Path, queue_size=1)
msg_fren = build_path_msg([pose[0], path_pose[0]], [pose[1], path_pose[1]])
pub_fren.publish(msg_fren)
```
This is a part of my code which I when running, I get an error as 

```log
:0: UserWarning: You do not have a working installation of the service_identity module: 'cannot import name verify_ip_address'.  Please install it from <https://pypi.python.org/pypi/service_identity> and make sure all of its dependencies are satisfied.  Without the service_identity module, Twisted can perform only rudimentary TLS client hostname verification.  Many valid certificate/hostname mappings may be rejected.
2019-07-17 17:50:15,202 [DEBUG] Started to connect...
2019-07-17 17:50:15,206 [DEBUG] Server connected: tcp4:127.0.0.1:9090
2019-07-17 17:50:15,206 [INFO] Connection to ROS MASTER ready.
S:  0 36.4417313996
RELOCATE
Traceback (most recent call last):
  File "ros_pp.py", line 215, in <module>
    pub_fren.publish(msg_fren)
  File "/usr/local/lib/python2.7/dist-packages/roslibpy/core.py", line 162, in publish
    'msg': dict(message),
TypeError: 'Path' object is not iterable
```

I think that the error is because I haven't converted the message to ROS2 format. I tried using `class roslibpy.Message(values=None)[source]` by ` msg_fren = roslibpy.Message(msg_fren)
` but it isn't helping either. Can you suggest me what is wrong in my code?

gonzalocasas, 2019-07-17:
Your constructor for Topic is wrong, message_type is a string, e.g. `sensor_msgs/CameraInfo`, but you're passing a python type (`Path`)

 Please check the docs: https://roslibpy.readthedocs.io/en/latest/reference/index.html#topics