RobotWebTools/roslibpy #33: Event for Connection Lost

RGring, 2019-08-07:
Hello people.
I connect to ROS Master via network and if the network connection is down, I would like to get an event or something.

Here it seems that it should throw an event 'close', but I is never called.
https://github.com/gramaziokohler/roslibpy/blob/master/src/roslibpy/comm/comm_autobahn.py#L89

In general it seems, that that a broken tcp-connection is not recognized be roslibpy. The state of factory.connector.state remains on 'connected'.

When the network connection is re-established it takes about ~3 min until I receive data from the topics again.

## System Information 
Ubuntu 16.04, ROS Kinetic

I hope you can help me to understand it better. :-)
Thanks in advance

gonzalocasas, 2019-08-07:
@RGring what exactly is the network failure mode you have? Is there a way to simulate the failure, e.g. unplug or turn off wifi and then reconnect causes this behavior?

RGring, 2019-08-09:
I get an event, when I shutdown the rosmaster. But I don't get an event, when I turn off the wifi connection, keeping the rosmaster running. But it seems that it reconnects very stable after the connection is re-established, although it takes some time.

gonzalocasas, 2020-03-25:
@RGring this should be fixed in 1.0. could you please confirm?

gonzalocasas, 2020-04-09:
I will close this, because I believe it should be resolved with the `1.0` release. @RGring if that is not the case, please reopen (or create a new issue if the behavior has changed but it is still not correct).

closed 2020-04-09 by gonzalocasas