RobotWebTools/roslibpy #34: Sporadically not connecting, Ros.run() hangs

HWiese1980, 2019-08-23:
Sometimes I cannot connect to the rosbridge. `Ros.run()` hangs forever or fails when a given timeout expires. The `rosbridge_server/rosbridge_websocket` is running, `roslibjs` is able to connect without issues. 

It seems like `on_ready` never fires.

## Expected Result
Connection establishing

## Actual Result
Connection fails sporadically

## Reproduction Steps
```python
import roslibpy

r = roslibpy.Ros(host="ws://localhost", port=9090)
r.run() # here it hangs or fails if given timeout expires
```

## System Information
Ubuntu 16.04 LTS, roslibpy 0.6.0 (running with Python 2.7 or within a Python 3.6 conda environment; doesn't make a difference, problem appears in both versions), ROS Kinetic (running with Python 2.7)

gonzalocasas, 2019-09-09:
This is most likely caused by an internal timeout of the `rosbridge`, which causes all sort of weirdnesses. A quick'n'dirty workaround is to set it to a very high value, e.g. launch rosbridge with the following parameter: `unregister_timeout:=28800`.

Could you please try that out and let me know if it improves the situation?

HWiese1980, 2019-09-24:
I haven't been able to reproduce the problem yet... as soon as I encounter it again (I hope that won't happen though), I'll give it a try. Thanks for the advice.

closed 2019-09-24 by gonzalocasas