RobotWebTools/roslibpy #35: can not get topics or server

Fimode, 2019-08-29:
i have a problem, i use get_topics() but return 'None',  i am sure it connected, when i use is_connect(), it returns True

gonzalocasas, 2019-09-09:
The issue was that the `get_topics` method (and other similar ROS API methods) were only async, so, they return the value inside the callback, not directly as the return value. This is now changed, so by default (i.e. if no callback is passed), the method will return directly the values as expected.

closed 2019-09-09 by gonzalocasas

gonzalocasas, 2019-09-09:
@Fimode just released `0.7.0` which fixes this behavior. Also added documentation of how to use the feature: https://roslibpy.readthedocs.io/en/latest/examples.html#querying-ros-api

commit 080c0e03 (2019-09-09): closes #35: Make it easier to work with the ROS API