RobotWebTools/roslibpy #36: `invoke test` stuck at very first test

HWiese1980, 2019-09-04:
`invoke test` stuck at `test_param.py`. I can see the test script client connecting to a running `rosbridge_websocket` server instance as it states an incoming connection. But after that, nothing more happens.

## Expected Result
Tests to continue

## Actual Result
Tests stuck right from the start

## Reproduction Steps

* launch `rosbridge_websocket` sever
* `invoke test`

## System Information
* Operating system: Ubuntu 16.04 LTS
* ROS Kinetic Kame
* Python: 3.6.9 (yes, rosbridge appears to work in this environment; might not officially be supported yet, not sure, but it should be, as Python 2.7 is EOL by the end of this year, just as a reminder)

For any further details feel free to ask!

gonzalocasas, 2019-09-04:
@HWiese1980 afaik, ROS1 does not really play well with Python 3 (regardless of the EOL of 2.x). To use Python 3, what we normally do is install Miniconda, create an environment with Python 3.x, and install `roslibpy` and so on in there, leaving the system Python 2.x alone for ROS to work fine.

Could you give that a try and let us know whether it works better that way?

HWiese1980, 2019-09-04:
Okay, this is what I'm currently getting in a clean Python 2.7 conda environment:

```
Checking MANIFEST.in...
lists of files in version control and sdist match
Checking ReStructuredText formatting...
running check
warning: Check: This command has been deprecated. Use `twine check` instead: https://packaging.python.org/guides/making-a-pypi-friendly-readme#validating-restructuredtext-markup

The project's long description is valid RST.
Running flake8 python linter...
Checking python imports...
ERROR: /home/user/my_repos/roslibpy/src/roslibpy/comm/comm_cli.py Imports are incorrectly sorted.
--- /home/user/my_repos/roslibpy/src/roslibpy/comm/comm_cli.py:before	2019-09-04 09:51:29.330989
+++ /home/user/my_repos/roslibpy/src/roslibpy/comm/comm_cli.py:after	2019-09-04 10:36:34.697700
@@ -23,9 +23,9 @@
 from System.Threading import Thread
 from System.Threading.Tasks import Task
 
+from ..event_emitter import EventEmitterMixin
 from . import RosBridgeException
 from . import RosBridgeProtocol
-from ..event_emitter import EventEmitterMixin
 
 LOGGER = logging.getLogger('roslibpy')
 RECEIVE_CHUNK_SIZE = 1024
ERROR: /home/user/my_repos/roslibpy/src/roslibpy/comm/comm_autobahn.py Imports are incorrectly sorted.
--- /home/user/my_repos/roslibpy/src/roslibpy/comm/comm_autobahn.py:before	2019-09-04 09:51:29.330989
+++ /home/user/my_repos/roslibpy/src/roslibpy/comm/comm_autobahn.py:after	2019-09-04 10:36:34.718461
@@ -12,8 +12,8 @@
 from twisted.internet.protocol import ReconnectingClientFactory
 from twisted.python import log
 
+from ..event_emitter import EventEmitterMixin
 from . import RosBridgeProtocol
-from ..event_emitter import EventEmitterMixin
 
 LOGGER = logging.getLogger('roslibpy')
```

gonzalocasas, 2019-09-04:
yeah, ok, but that's a styling issue; are you trying to develop `roslibpy` further, i.e. contribute directly to the project? or are you intending to use it in your setup? `invoke test` is for the development workflow, not for the user's workflow. If you want to simply run the tests, for curiosity's sake, then run `pytest` (that will be almost the same as `invoke test` but exclude all the code style checks and other stuff that we run before releases).

HWiese1980, 2019-09-04:
Yes, I'm actually trying to do that. I have started writing `message_filter` like topic synchronizers for the library that I'd like to contribute.

HWiese1980, 2019-09-04:
Or, lets say, I've written them independent of the library and would like to offer them to be added.

gonzalocasas, 2019-09-04:
Oh cool! Let me know if you managed to get the tests to work or if you need help for that. I'll fix the tiny styling errors you got earlier with `invoke test`

gonzalocasas, 2019-09-09:
@HWiese1980 the styling issues are now fixed: https://github.com/gramaziokohler/roslibpy/commit/3a05727c9351dba3fc9aedb2fb44ee5cb852311d :)

I'm going to close this issue, as it seems it is fixed; if you are still having (other) issues when trying to setup the environment to contribute, please feel free to keep commenting here, or create a new issue.

closed 2019-09-09 by gonzalocasas