RobotWebTools/roslibpy #38: Data type compatibility while sending images over rosbridge

aakashsehgal, 2019-09-18:
I receive a video stream from an environment simulation tool. I reshape the image, convert it into uint8 and then publish it. 

```
ros = roslibpy.Ros(host='localhost', port=9090)
ros.on_ready(lambda: print('Is ROS connected ?', ros.is_connected))
ros.imagePublisher = roslibpy.Topic(ros,'/camera/image/compressed','sensor_msgs/CompressedImage')

i = np.array(image.raw_data)
i2 = i.reshape((480, 680, 4))
i3 = i2[:, :, :3]
imageProcessed = i3.astype(np.uint8)
ros.imagePublisher.publish(roslibpy.Message({'format' : "jpeg", 'data' : imageProcessed}))

ros.run_forever()
```
But this gives an error `Failed to send message, Object of type ndarray is not JSON serializable`. After looking at [this](https://stackoverflow.com/questions/51685322/object-of-type-ndarray-is-not-json-serializable/51685601) post, I converted it into a list by `imageProcessed = i3.astype(np.uint8).tolist()`.

Now I can publish the image and also see this topic on the ros side. The problem is that I cannot see the contents of this message because ros gives an error : `[ERROR] [1568804662.572294]: [Client 16] [id: publish:/camera/image/compressed:9] publish: field data[] must be unsigned integer type`

What could be done in this case ? 

Republishing the image to a normal `/sensor_msgs/image topic` can also not be done since the message type is not uint

gonzalocasas, 2019-09-18:
Don't you need to re-encode the image into base64 before you publish it?

aakashsehgal, 2019-09-18:
No. Why would it be required ? According to the message definition of `/camera/CompressedImage` , i need to transfer an array of unint8. Isnt it ?

gonzalocasas, 2019-09-18:
JSON in general doesn't like byte arrays too much, so, `rosbridge` uses base64 encoding to turn them into an ascii-only representation of the byte array (or `uint8[]` in this case), and handles the decoding on the other side. `roslibpy` does not add any additional processing (at least for now, in the future, I think we will add this directly to the library), so I expect you need to pass it as a base64-encoded string for it to work.

aakashsehgal, 2019-09-18:
Do you mean something like this ?

```
i3Str = json.dumps(i3)
base64EncodedStr = base64.b64encode(i3Str)
ros.imagePublisher.publish(roslibpy.Message({'format' : "jpeg", 'data' : base64EncodedStr})
```

I looked up on the internet...this also gives the same error : `Object of type ndarray is not JSON serializable`

gonzalocasas, 2019-09-18:
No, that error is telling you that line one (`json.dumps(i3)`) is impossible, because you cannot jsonify a numpy array. You don't need to do any `json.dumps` thou, you need to take that array (which you should not need to cast to unit8 array, an ndarray should be fine) and base64-encode that, so, something like:
```
base64EncodedStr = base64.b64encode(i3)
ros.imagePublisher.publish(roslibpy.Message({'format' : "jpeg", 'data' : base64EncodedStr})
```

aakashsehgal, 2019-09-18:
thanks for the hint, however this further gives another error :` ValueError: ndarray is not C-contiguous`
I tried to use `np.ascontiguousarray()` but this also does not help.

aakashsehgal, 2019-09-18:
I used `np.array2string(i3)`. This works. But again I need to use string2array() on the ROS side to decode it.

aakashsehgal, 2019-09-20:
I guess I found the problem while using `base64EncodedStr = base64.b64encode(i3)`. This gives an error : `ValueError: ndarray is not C-contiguous` because the size of the image is too big and it cannot encode such a big array to a string. 
I tested the same functionality on a 7Kb image.
```
image = cv2.imread('test_image_lr.jpg')
print(image.shape)

encoded = base64.b64encode(image)
print('encodedString', encoded)
```
and the output is:  

```
(114, 208, 3)
encodedString Squuezed text (1186 lines)
```
So basically the encoded string is 1186 lines long and that is only for a 7kb image...for a video it simply does not work since the size is way too large.

Can we solve this in some other way ?

gonzalocasas, 2019-09-20:
Did you try the solution here? https://stackoverflow.com/questions/43310681/how-to-convert-python-numpy-array-to-base64-output
```
pil_img = Image.fromarray(img)
buff = BytesIO()
pil_img.save(buff, format="JPEG")
new_image_string = base64.b64encode(buff.getvalue()).decode("utf-8")
```

Btw, since these problems are actually unrelated to the `roslibpy` codebase, I'm going to close this issue. However, feel free to keep commenting on it.

closed 2019-09-20 by gonzalocasas