RobotWebTools/roslibpy #40: Use roslibpy in conjunction with asyncio instead of twisted?

HWiese1980, 2019-09-25:
I'm using `asyncio` in one of my projects. `roslibpy` on the other hand seems to use `twisted` which is fine, however it looks like I cannot use both. 

I would like to create an `autobahn` websocket server apart from `roslibpy` using `asyncio`. The server itself works but `roslibpy` complains afterwards that `twisted` is `Explicitly using 'asyncio' already`. That makes sense but is blocking me. 

Is it not possible to use both, `twisted` for `roslibpy` and `asyncio` for another websocket server in the same application? 

## Expected Result
`asyncio` and `twisted` should work alongside each others. 

## Actual Result
Error message: `Explicitly using 'asyncio' already`.

## Reproduction Steps

* Set up a project that opens a `roslibpy` connection to a `rosbridge_websocket` server
* Create another websocket server by means of `autobahn`'s `WebSocketServerFactory` using a protocol derived from `autobahn.asyncio import WebSocketServerProtocol`

## System Information
* Ubuntu 16.04 LTS (with Gnome)
* `roslibpy` 0.7.0
* `autobahn` 19.9.3
* `twisted` 19.7.0

gonzalocasas, 2019-09-25:
@HWiese1980 the current implementation of the `autobahn` connector uses `twisted`, but `roslibpy` is designed to have swappable comm layers (there's an alternative layer already that uses .NET web sockets for use inside IronPython).

It would be doable (even relatively easy) to create a new one that uses `autobahn`'s `asyncio` components instead. If you'd have time to experiment with it, it would be great if you would be up to send a PR to add that..!

gonzalocasas, 2022-09-17:
@HWiese1980 excuse the very late response, but this can be easily solved with `asyncio.run` (python 3.7+), eg.:

```
async def connect_async_client(address):
    client = BleakClient("127.0.0.1")
    await client.connect()
    return client

if __name__ == "__main__":
    import roslibpy
    client = RosClient()
    some_other_async_client = asyncio.run(connect_async_client())

    service = roslibpy.Service(client, '/do_something', 'srv/ServiceType')
    service.advertise(functools.partial(handler, some_other_async_client=some_other_async_client))

    try:
        client.run_forever()
    except KeyboardInterrupt:
        service.unadvertise()
        client.terminate()
```

closed 2022-11-23 by gonzalocasas