RobotWebTools/roslibpy #41: Increasing delay with queue_size/queue_length = 1 or 0, and long running subscriber callback

HWiese1980, 2019-10-09:
I'm experiencing the following weird and undesirable behavior: messages are increasingly long delayed when the subscriber callback takes too long to react on the following message, even if queue_length/queue_size are set to 1 or to 0. 

Below I've attached an example script to reproduce the behavior. 

## Expected Result
Age of messages does not increase, only the newest message is delivered

## Actual Result
It seems like the delivered messages get older and older, as if they are infinitely queued somewhere on the way. I can see this behavior by sending `std_msgs/Header` messages with the current time. I'm receiving messages, measuring their age by subtracting their time stamp from the current time. The result is an increasing "age" of the currently processed message. 

I've tested the same with `rospy` and can only reproduce the behavior there if I leave the `queue_size` of the subscriber on default. Also `buff_size` seems to play a role .

I have already modified the `subscribers.py` of `rosbridge_server` accordingly (added queue_size = 1, buff_size = 2**24) but that unfortunately has not changed anything about the behavior.

## Reproduction Steps
```python

import roslibpy
import time
import json
import logging
logger = logging.getLogger("roslibpy")
handler = logging.StreamHandler()
logger.addHandler(handler)

def to_epoch(stamp):
    stamp_secs = stamp["secs"]
    stamp_nsecs = stamp["nsecs"]
    return stamp_secs + stamp_nsecs*1e-9

def from_epoch(stamp):
    stamp_secs = int(stamp)
    stamp_nsecs = (stamp - stamp_secs) * 1e9
    return {"secs": stamp_secs, "nsecs": stamp_nsecs}

def sub(msg):
    # The `age` of the message should remain constant, but it increases seemingly indefinetly
    age = time.time() - to_epoch(msg["stamp"])
    print("Age of message: %6.3f seconds" % age)
    time.sleep(1./5.) # increase if necessary; above a certain loop time the described behavior occurs 

ros = roslibpy.Ros("ws://localhost:9090")
ros.run()

tt = roslibpy.Topic(ros, "/test", "std_msgs/Header", queue_size=1)

ts = roslibpy.Topic(ros, "/test", "std_msgs/Header", queue_length=1)
ts.subscribe(sub)

try:
    seq = 0
    while True:
        seq += 1
        header = {"frame_id": "test", "stamp": from_epoch(time.time()), "seq": seq}
        tt.publish(roslibpy.Message(header))
except KeyboardInterrupt:
    ros.terminate()

```

## System Information
Ubuntu 16.04 LTS, ROS Kinetic Kame

gonzalocasas, 2019-10-15:
@HWiese1980 this is very interesting, thanks a lot for the detailed repro! It's also good to know it's probably related to `rosbridge` and not this library, but I will look into ways to mitigate this.

HWiese1980, 2019-10-21:
Awesome! Would be cool if you could keep me updated.

chia-labradorsystems, 2020-03-06:
I'm experiencing the same issue as well.
The issue still exists on the latest ```rosbridge``` develop branch

HWiese1980, 2020-03-06:
If I remember correctly the problem is a low-level queue somewhere in the network stack. I don't think it can be solved while sticking with websockets. What I've been wondering about is that even plain UDP packet based rosbridge is affected. I would expect UDP not to have a queue. If there isn't a receiver for a UDP packet, I'd expect it to be dropped instead of queued. But I'm not a network expert (anymore... it's been almost 20 years now since I learned that stuff). 

I had been working on an MQTT based ROS bridge server before I left my previous employer. It isn't really sophisticated but worked. Can't really remember though if I had experienced that increasing delay there too... I don't think so, but I'm not sure. Take a look at my rosbridge_suite fork if you like. It's among the server scripts: https://github.com/HWiese1980/rosbridge_suite

gonzalocasas, 2020-03-27:
I've been analyzing this in depth today using WireShark to understand where the problem might be, and I don't think it's an issue of the underlying layers, nor of the `rosbridge_suite`.

I prepared a little script (very similar to the one posted by @HWiese1980 on the description) that reproduces the drifting issue on Windows. But then, when I run the exact same script using IronPython instead of CPython there is no delay at all.

When `roslibpy` is used from IronPython, it relies on a different websockets implementation based on .NET `ClientWebSocket`, and when used from CPython, it uses one based on Autobahn. 

I seriously doubt the issue on `autobahn`, so I expect we're not setting some of its parameters correctly, or we're not handling some event the way autobahn expects us to do, which causes this problem. I will keep investigating, but in principle, the bug is isolated to this file: https://github.com/gramaziokohler/roslibpy/blob/master/src/roslibpy/comm/comm_autobahn.py

gonzalocasas, 2020-04-01:
The underlying problem seems to be linked to the event loop.

At least on Windows, we chased down the drifting behavior to this bit of code in twisted's `win32select`, which has a 500ms default timeout for `select` calls: https://github.com/twisted/twisted/blob/trunk/src/twisted/internet/selectreactor.py#L36
If the inlined comment is correct, this is added only to keep CLI apps responsive to ctrl+c aborts.

However, I am not entirely sure why the `select` operation is leaving via timeout instead of triggered by its normal code path, but a quick and dirty fix is to add the following two lines at the start of your code:
```
from twisted.internet import reactor
reactor.timeout = lambda : 0.0001
```

This forces the timeout used by `select` to be really small, and seems to put down the latency to a max of 4ms on my local setup (for reference, without this fix, the latency keeps increasing from a few ms up to 500ms, and then drops and repeats).

gonzalocasas, 2020-04-01:
@HWiese1980 I would be interested in hearing your feedback about my previous comment, since you use Ubuntu (according to your initial report) instead of linux, this might have a different effect, but I expect it to also "fix" the observed problem, with the only difference than you might not have the 500ms default timeout used for Windows.

HWiese1980, 2020-04-01:
@gonzalocasas Happy to hear that you can reproduce it and have been able to pin down a pretty specific probable cause. I'm unfortunately currently not on any project anymore that deals with this issue because I switched my employer meanwhile but I can try to test it on my private PC. I'll have to set it up for this purpose though. I suppose I'll first test it on Windows again inside a virtual machine and/or WSL2 (which I only got because I've  joined the MS Insider Program) and if that does not yield reliable and credible results I'll try it on a dual boot Ubuntu which is flying around on one of my harddisks. I'll report back as soon as I know more.

yck011522, 2020-04-01:
@gonzalocasas  Thanks a lot for the fix. 🙇 

I had this problem before with my **Windows + VMware Ubuntu setup**. And can reproduce the problem as you described. **delay up to 500ms**.

After your _2 lines of code_, I am able to get a much lower delay **down to 0 - 50ms range** (more often ~10 - 20ms) . It is not a systematic test that I did, but observing from the log file, it looks much better. 

I use **tkinter** as UI and the **roslibpy** on a separate thread. I observed no side effect so far regarding non-responsiveness or anything similar.

HWiese1980, 2020-04-01:
@gonzalocasas I've tried your workaround with my testing code (as can be seen in my initial post) and tested it on a WSL 2 based Ubuntu 18.04 LTS on Windows 10. Unfortunately in this configuration nothing changes. The result is still the same. Although it seems a different result than you have. My delay is not just growing towards 500ms and dropping back to 0. It's constantly growing, producing the following output on all the systems I've tested this on:

```
Age of message:  0.283 seconds
Age of message:  0.598 seconds
Age of message:  0.833 seconds
Age of message:  0.974 seconds
Age of message:  1.286 seconds
Age of message:  1.508 seconds
Age of message:  1.665 seconds
Age of message:  1.979 seconds
Age of message:  2.133 seconds
Age of message:  2.433 seconds
Age of message:  2.669 seconds
Age of message:  2.827 seconds
Age of message:  3.142 seconds
Age of message:  3.377 seconds
Age of message:  3.592 seconds
```

It also doesn't matter where I put the two lines (reactor.timeout = lambda...) you suggested as workaround above. 

I'm not quite sure how much the WSL 2 environment does affect this and still have to try it in an actual VM and on a real Ubuntu system. Will report back once I've new results.

gonzalocasas, 2020-04-02:
@HWiese1980 thanks for testing again! I just ran your script and got the exact same results as you, but on closer inspection, I believe part of the problem is the testing script itself. You are publishing at full, unthrottled speed (you `publish` loop has no waiting at all), and your throttling is done on the subscribing callback. This callback is not handled on a separate thread as you might expect, so the sleep inside the `sub` method affects your publishing as well because it blocks the event loop.

To fix the script, you need to first move the `time.sleep(1./5.)` outside of your `sub` callback and put it right after the `tt.publish(roslibpy.Message(header))` in the publishing loop. And once that is done, the `reactor.timeout` fix will work.

So, here are my results, with your script untouched:
```
Age of message:  0.647 seconds
Age of message:  0.858 seconds
Age of message:  1.071 seconds
Age of message:  1.298 seconds
Age of message:  1.472 seconds
Age of message:  1.732 seconds
Age of message:  1.943 seconds
Age of message:  2.155 seconds
Age of message:  2.398 seconds
Age of message:  2.606 seconds
Age of message:  2.818 seconds
Age of message:  3.030 seconds
Age of message:  3.242 seconds
Age of message:  3.423 seconds
Age of message:  3.665 seconds
Age of message:  3.877 seconds
Age of message:  4.090 seconds
Age of message:  4.314 seconds
Age of message:  4.524 seconds
```

If I move the sleep from the sub to the publishing loop, the same behavior of going up to 500ms and back to 0ms shows up:
```
Age of message:  0.402 seconds
Age of message:  0.202 seconds
Age of message:  0.003 seconds
Age of message:  0.306 seconds
Age of message:  0.107 seconds
Age of message:  0.410 seconds
Age of message:  0.211 seconds
Age of message:  0.011 seconds
Age of message:  0.327 seconds
Age of message:  0.127 seconds
Age of message:  0.431 seconds
Age of message:  0.231 seconds
Age of message:  0.031 seconds
Age of message:  0.334 seconds
Age of message:  0.135 seconds
```

Finally, if I apply the `reactor.timeout` fix, the latency goes down to the expected single-digits ms range:
```
Age of message:  0.003 seconds
Age of message:  0.002 seconds
Age of message:  0.002 seconds
Age of message:  0.004 seconds
Age of message:  0.001 seconds
Age of message:  0.001 seconds
Age of message:  0.002 seconds
Age of message:  0.002 seconds
Age of message:  0.003 seconds
Age of message:  0.002 seconds
Age of message:  0.003 seconds
Age of message:  0.002 seconds
```

Could you please confirm you see the same on your side? Thanks a lot!!

gonzalocasas, 2020-04-02:
@HWiese1980 @yck011522 one more comment on this: part of the problem is that the while-loop for publishing on both your test scripts is not really friendly to the way multitasking works on python/twisted. If you invert the loop and write it using the `call_later` behavior instead, you don't need to use the workaround of a low timeout on the reactor. eg:

```
ros = roslibpy.Ros("ws://localhost:9090")

tt = roslibpy.Topic(ros, "/test", "std_msgs/Header", queue_size=1)
ts = roslibpy.Topic(ros, "/test", "std_msgs/Header", queue_length=1)
ts.subscribe(sub)

seq = 0
def send_msg():
    global seq
    seq += 1
    header = {"frame_id": "test", "stamp": from_epoch(time.time()), "seq": seq}
    tt.publish(roslibpy.Message(header))
    ros.call_later(1, send_msg)

ros.run_forever()
```

HWiese1980, 2020-04-03:
I've created a testing/demo repository for my experiments regarding this topic. I'm still experiencing the increasing delay. You can find the code in the following repo: https://github.com/HWiese1980/roslibpytest. 

I've also incorporated the fixes that you, @gonzalocasas, suggested (`reactor.timeout` and `ros.call_later`), but it did not change anything about the observes behavior. At least not on my WSL 2 Ubuntu. I still have to test it on a real Ubuntu machine. But since WSL 2 uses a real Linux kernel I'm not sure if it will make a difference. We'll see...

gonzalocasas, 2020-04-03:
@HWiese1980 thank you very much for the super detailed preparation! I see one problem with the new script thou. It still has the time.sleep inside the subscriber callback, that will definitely cause issues: https://github.com/HWiese1980/roslibpytest/blob/master/scripts/test_sub.py#L20

Also, I would suggest to remove the `ros.run()` line in `test_sub.py` and add a `ros.run_forever()` at the end instead of the busy loop.

HWiese1980, 2020-04-03:
The `time.sleep` in the subscriber simulates a very long running subscriber callback that takes longer than the period of the incoming messages, and it simulates the actual trigger of the observed behavior. I've originally had a program that processed data that came in at a high frequency. The processing took quite long. So I built everything in a way that the subscriber callback only put the data in a global variable and did nothing more. Another periodically executed method then processed the latest data in that global variable. While it was running, new data was dropped and repeatedly overwritten until it was ready to process the newst latest data. The problem persisted because the program was too busy processing the data to switch over to running the subscriber callback. And so data came in faster than it could be processed which lead to the increasing age. My understanding of the `queue_length` parameter is that basically like, if I set it to 1, there can be only one message in the queue and if a new one comes in, **the older one** gets dropped and replaced by the new one (as what I did manually as described above). However in practice it looks like **newer** data gets dropped instead of being enqueued and that the message currently in the queue (of length 1) always stays there until a subscriber callback processes it. 

Aside from that I've now replaced `ros.run()` by `ros.run_forever()`.

gonzalocasas, 2020-04-03:
Ah, but that will never work as you expect then. Python is (for the most part) a single-threaded language, so, if you consumer takes longer than the production, it will always lag behind.

You will need to resort to another way to distribute the computation, build a way to preempt processing or ignore messages if they are too old, or something like that. On one extreme of the spectrum, you might need to spin a full process per message that arrives if the computation really takes very long, but there are surely more efficient ways to split/distribute that load.

HWiese1980, 2020-04-03:
Well, I'm sorry, but my expectation might be off then... I'd have expected a queue of length 1 to behave more like a variable that contains only the *newest* available data, not the one that came after the one currently being processed (except on the very first run, when nothing's being processed; see further down). I don't really see how the way I observe it to behave, can be a desireable behavior in robotics. I mean, for example, if it's visual data, an image that requires some processing time (deep learning, I don't know), I'd expect the next sensor data frame to be dequeued for processing to be the almost *newest* data available. For instance, if I have an autonomous robot, a car or whatever, and it drives along some road (other traffic neglected) and it sees a road sign, needs some time to process it, let's say, 5 seconds, I would not want it, after it finished with the frist image, to process the image that came in right after the previously processed one because that's already almost 5 seconds (minus image data period time) old. I'd want the one the camera saw most recently, ideally right now. 

My expected behavior of a queue of length 1 goes as follows: at the beginning, the very first item that comes in (let's call it 'A') is enqueued and immediately dequeued for processing. The one that follows immediately on the current one ('B') is then enqueued until processing of A has finished. While A is being processed and B is waiting, C, D, E, F get dropped because the queue is full. After A is finished, B gets picked and G is enqueued. While B is being processed and G is sitting in the queue, H, I, J and K get dropped because no room in the queue. G is then picked for processing and, since there's room again in the queue, L is enqueued... and so on... 

If this is correct, I would expect the latest dequeued item to be at max as old as my processing time because it's the first one that comes in from the sensor after the queue is empty again. So how is it possible that the items that I'm dequeueing get older and older when they cannot be queued because the queue of length 1 is full (one item waiting) until my processing is done? There has to be some other queue before the queue of length 1 for this to happen. 

Or am I completely mistaken? I'm not quite sure right now...

gonzalocasas, 2020-04-03:
Oh! Now I see what you mean: You want to **drop packages** that cannot be processed in time by a slow consumer. This can be achieved, but you're missing a parameter: `throttle_rate` is required to be used in combination with `queue_length` for it to work as you expect. The queueing happens entirely on `rosbridge_server` not on the client side, that's why you could reproduce this using `roslibjs`.

In summary, you need to set both `queue_length=1` and `throttle_rate` tuned to the duration of your slow consumer. Notice than despite the `rate` word in the name, this parameter is specified in milliseconds, not hz.

The following script (based on yours) simulates the behavior you want:

```python
#!/usr/bin/env python
import roslibpy
import time
import logging

logger = logging.getLogger("roslibpy")
handler = logging.StreamHandler()
logger.addHandler(handler)

def to_epoch(stamp):
    stamp_secs = stamp["secs"]
    stamp_nsecs = stamp["nsecs"]
    return stamp_secs + stamp_nsecs*1e-9

def from_epoch(stamp):
    stamp_secs = int(stamp)
    stamp_nsecs = (stamp - stamp_secs) * 1e9
    return {"secs": stamp_secs, "nsecs": stamp_nsecs}

def sub(msg):
    age = time.time() - to_epoch(msg["stamp"])
    print("Age of message (#%d): %6.3f seconds" % (msg["seq"], age))
    # Simulate a very slow consumer
    time.sleep(2)

ros = roslibpy.Ros("ws://localhost:9090")

tt = roslibpy.Topic(ros, "/test", "std_msgs/Header")

# Queue length needs to be used in combination with throttle rate (defined in MS!).
# This value must be tuned to the expected duration of the slow consumer
# and ideally bigger than the max of it, otherwise message will be older than expected (up to a limit)
ts = roslibpy.Topic(ros, "/test", "std_msgs/Header", queue_length=1, throttle_rate=2200)
ts.subscribe(sub)

try:
    seq = 0
    def send():
        global seq
        seq += 1
        header = {"frame_id": "test", "stamp": from_epoch(time.time()), "seq": seq}
        tt.publish(roslibpy.Message(header))
        ros.call_later(.001, send)

    ros.on_ready(send)
    ros.run_forever()
except KeyboardInterrupt:
    ros.terminate()
```

And the output, as expected, is that age is relatively constant and there are gaps in the sequence of received messages:
```
$ python github_test.py
Age of message (#39):  0.012 seconds
Age of message (#160):  0.003 seconds
Age of message (#272):  0.004 seconds
Age of message (#389):  0.003 seconds
Age of message (#508):  0.004 seconds
Age of message (#628):  0.003 seconds
Age of message (#748):  0.004 seconds
Age of message (#868):  0.003 seconds
Age of message (#988):  0.003 seconds
Age of message (#1106):  0.004 seconds
Age of message (#1223):  0.003 seconds
Age of message (#1344):  0.002 seconds
Age of message (#1455):  0.003 seconds
Age of message (#1581):  0.004 seconds
Age of message (#1699):  0.010 seconds
Age of message (#1826):  0.002 seconds
Age of message (#1959):  0.002 seconds
Age of message (#2095):  0.005 seconds
Age of message (#2224):  0.007 seconds
Age of message (#2359):  0.003 seconds
Age of message (#2496):  0.003 seconds
Age of message (#2609):  0.004 seconds
Age of message (#2721):  0.005 seconds
Age of message (#2846):  0.003 seconds
```

I also tested the same but splitting publisher and subscriber scripts, just to be sure the measurement is not affected by the fact both are in the same process, and it also seems to work.

gonzalocasas, 2020-04-03:
>  They must be queued somewhere else too, or they wouldn't age like that.

Could you elaborate what you mean by that? As far as I could see (and visible in in the output posted in my previous comment), the packages have consistently a very low age, meaning we're always getting the very last one, which happens to be very recent all the time.

> Also packages are being dropped somewhere on the way already, as the packages don't come in sequentially. There are gaps in the sequence.

Also not sure I get this. The packages are dropped by the `rosbridge` by setting `queue_length` and `throttle_rate`. They come sequentially (as in, always in-order), but of course there are gaps, because that is the intended behavior: you said you wanted packet drop to occur, so, consequently, there are gaps in the sequence. Did I miss something?

HWiese1980, 2020-04-04:
I'm sorry, I'm still talking about the case without `throttle_rate` where the publisher pumps out what it has. I'm trying to understand what's happening there and why the age is increasing. 

If I publish a sequence A, B, C, D, E, F at a rate of one item per second and process it at a rate of lets say one item in (slightly more than) three seconds with a queue size of one, I'd expect the following to happen:

A will be published/enqueued and immediately received and dequeued on subscriber side. B will be enqueued one second after A. A's processing takes three seconds. So while A is processed and B waits in the queue, C gets produced but cannot be enqueued because B is still in the queue. So it is dropped. Then D gets produced and also dropped because B is still waiting in the queue for A to be finished. A is then finished, B gets dequeued and its processing starts. E is produced and enqueued. So B's age when it's being processed is about 2 seconds. While B is being processed, F and G are produced but dropped because E is waiting. When B is finished, E is dequeued and it's age will be around 3 seconds. H will find a place in the queue. While E is processed, I and J will be dropped and when E is finished, H will be dequeued and it's age will be around 3 seconds, not increasing any further than the time the processing of one item takes (3 seconds).

What really happens though, for some reason that I don't understand yet but would love to, is that the age of the message increases far beyond the processing period. That's why I reckon that there must be some kind of additional queue somewhere where those messages that are far older than the processing time can remain in - but only some of them, as it seems, because even without `throttle_rate` I don't seem to receive all messages. 

That's what I'm currently after. The solution of this issue might indeed be to use the `throttle_rate` parameter. So feel free to close the issue, but if you like to see where this phenomenon comes from, let's keep investigating. I'm really curious to see.

gonzalocasas, 2020-04-04:
> I'm still talking about the case without `throttle_rate` where [..]

Ah! Ok. The case without `throttle_rate` is easier to understand: without the parameter, the `queue_length` parameter is not doing anything. See to the [`rosbridge` protocol](https://github.com/RobotWebTools/rosbridge_suite/blob/master/ROSBRIDGE_PROTOCOL.md#344-subscribe) spec that indicates that `**queue_length** – the size of the queue to buffer messages. Messages are buffered as a result of the throttle_rate`.  

So, without queueing, the results does not follow the sequence you just described. Instead, no message is dropped of any queue, it simply overwhelms the python side which cannot keep up since it handles all this in a single-threaded cooperative multitasking mode, not in true multi-threaded mode.

Notice also that the queueing behavior you specified where the older message stays in the queue and new ones are dropped, is not what happens either with the throttle rate enabled. According to the protocol spec, older messages are removed and replaced by newest message:

> If queue_length is specified, then messages are placed into the queue before being sent.
> Messages are sent from the head of the queue. If the queue gets full, the oldest message is removed and replaced by the newest message.

I am going to close this issue, because there is no action needed on `roslibpy` code at the moment. However, you're more than welcome to keep discussing and investigating the underlying phenomenon in here, even if closed! Thanks for your engagement!!

closed 2020-04-04 by gonzalocasas

gedeon1976, 2021-10-12:
This article was quite interesting, I was experimenting some kind of message aging for compressed image transmission on the following setup:

publisher
ROS2 foxy on a raspberry Pi 4B + 4G RAM

Bridge
ros2bridge with node.js

clients using roslibjs:
Surface 4 Laptop
S21 smartphone

connection via Wifi + CycloneDDS on both sides, reception buffer size = 10MB

Before adding the two parameters queue_length and throttle_rate the messaging was increasing by amount of seconds each every running second!

Adding the next values on the javascript client code produce a good behavior with apparently no message aging, other values show message aging for the S21 client.

```
//  Subscribe to Left image topic
            var imageL = new ROSLIB.Topic({
              ros : ros,
              name : '/left_stream/compressed',
              messageType : 'sensor_msgs/CompressedImage',
              // Add this two values to avoid message aging/increasing delay effect
              // check, https://github.com/gramaziokohler/roslibpy/issues/41
              // https://github.com/RobotWebTools/rosbridge_suite/blob/ros2/ROSBRIDGE_PROTOCOL.md#344-subscribe
              throttle_rate: 2200,
              queue_size: 10
             });
```

beasteers, 2023-01-28:
Thanks for the discussion on this.

The problem with the prescribed solution is that it assumes the publisher/consumers are running at a consistent & predictable rate, but that isn't always the case, especially when a device is being stressed. 

So if the only way to avoid processing old messages is to set the throttle rate, then that means that you'd have to set the throttle rate to the lowest FPS you might expect, but that means you're limiting your throughput to the worst-case scenario. And not only that, you're running the risk of your system breaking if the FPS ever shifts.

As an example, let's say you have a publisher that publishes data at anywhere from 15-1 fps (depending on what else is happening on the device), and you have a processing pipeline that runs between 4-10 fps and you want to visualize the data at full resolution (meaning you don't want to throttle the publisher). You'd need to throttle the consumer with 250ms to avoid lag in the worst case scenario and would never be able to run faster than that. And if you ever run slower than 4fps, you're going to start falling behind again.

In an ideal world, one shouldn't need to tune specific rates and it would work like @HWiese1980's "sequence A, B, C, D, E, F", (though a slightly more desirable way of it working IMO would be - when new items are produced, they push old items out of the other end of the queue like `collections.deque`, but I guess that's a bit trickier with network queues?)

I'm wondering if being able to select ROS's QOS parameters for publisher/subscriber could be a solution to these types of problems:

for example these looks promising ([source](https://docs.ros.org/en/humble/Concepts/About-Quality-of-Service-Settings.html#:~:text=Lifespan,effectively%20never%20received))

> History
> - Keep last: only store up to N samples, configurable via the queue depth option.
>
>Depth
> - Queue size: only honored if the “history” policy was set to “keep last”.
>
>Reliability
> - Best effort: attempt to deliver samples, but may lose them if the network is not robust.
>
>Durability
> - Volatile: no attempt is made to persist samples.
>
> Lifespan 
> - Duration: the maximum amount of time between the publishing and the reception of a message without the message being considered stale or expired (expired messages are silently dropped and are effectively never received).

But from what I can tell, they aren't configurable through rosbridge's protocol - which leads me to wonder, why aren't they?

These issues would happen with regular websockets as well, which is why I was so interested in ros because I thought it would offer a solution to this, but now I'm not sure. Is there a fundamental physical limitation that prevents functionality like this?

@gonzalocasas I'm curious if you have thoughts or insights into this? I get that this can't be resolved in roslibpy, but this was the most on-point discussion that I've seen.