RobotWebTools/roslibpy #42: Issue with custom Action

broidaj, 2019-10-14:
I am trying to send custom actions using roslibpy to an action server running regular ros. For some reason, the action goals aren't being published even though I'm getting no errors. My setup:

client = roslibpy.Ros(host='localhost', port=9090)
client.run()

action_client = roslibpy.actionlib.ActionClient(client, "action_planner", 'my_acts/CustomAction')
goal = roslibpy.actionlib.Goal(action_client, roslibpy.Message({'custom_data': data}))
goal.send()
result = goal.wait()

I've double checked that I can communicate with the server  by using normal ROS messages, and I've run the roslibpy action server examples, but for some reason when I switch to the custom messages it simply won't publish. I'm monitoring the goal topic manually and still don't see anything being published, even though I'm getting no errors. Any thoughts on the cause/a potential fix? I'm using ROS Melodic on Ubuntu 18.04

gonzalocasas, 2019-10-15:
Does your action server work if you just it from inside ROS instead of over the rosbridge/roslibpy? Could you check if the topic for the action goal result is published (using `rostopic echo`) ?

broidaj, 2019-10-15:
I tested it and it does work from inside ROS normally. The topic is created and I've been monitoring with rostopic echo, but when I use the rosbridge nothing gets published to it.

gonzalocasas, 2019-10-15:
@broidaj could you please post more details of your action? perhaps there's something with the types that fails to serialize?

broidaj, 2019-10-15:
Here is the action content

# Goal Section

trajectory_msgs/JointTrajectoryPoint joint_target

---
# Result Section

trajectory_msgs/JointTrajectory trajectory

---
# Feedback Section

# Nothing that I currently want here
# Didn't make this a service because planning can take a while
# so I want the potential interrupt capabilites of actions

gonzalocasas, 2019-10-15:
Would it be possible for you to define something in the feedback section, in order to test if the fact that there's nothing breaks something in the serializer somewhere?

broidaj, 2019-10-16:
Sure, I'll test it and get back to you.

gonzalocasas, 2019-11-15:
Any news? Should we close this?

gonzalocasas, 2020-01-12:
I am closing this due to inactivity, but if you still see the issue, please do re-open it.

closed 2020-01-12 by gonzalocasas