RobotWebTools/roslibpy #44: How to reconnect to rosbridge?

Airyzf, 2019-11-21:
How to reconnect to rosbridge? roslibpy auto reconnect ?

chia-labradorsystems, 2020-03-06:
roslibpy usually auto reconnects  if it is network issue.
But It seems like it cannot reconnect properly if rosbridge restarted.

This is how I test this issue:
First I connect to ROS successfully, then I restart rosbridge

roslibpy will prompt:
```
[139990187542272] 2020-03-05 16:02:06,799 [INFO] WebSocket connection closed: Code=1006, Reason=connection was closed uncleanly (peer dropped the TCP connection without previous WebSocket closing handshake)
[139990187542272] 2020-03-05 16:02:06,799 [DEBUG] Lost connection. Reason: [Failure instance: Traceback (failure with no frames): <class 'twisted.internet.error.ConnectionDone'>: Connection was closed cleanly.
]
[139990187542272] 2020-03-05 16:02:06,800 [INFO] Stopping factory <roslibpy.comm.comm_autobahn.AutobahnRosBridgeClientFactory object at 0x7f5202998d90>
```

Then I try to reconnect by using `connect()`

roslibpy prompt:
```
[139990196197120] 2020-03-05 16:03:14,493 [INFO] Starting factory <roslibpy.comm.comm_autobahn.AutobahnRosBridgeClientFactory object at 0x7f5202998d90>
[139990196197120] 2020-03-05 16:03:14,494 [DEBUG] Started to connect...
````

Everything seems fine except it never connects again

rosbridge server seems fine, once I restart the roslibpy client it connects again.
I've tested rosbridge with both the tornado version and the new autobahn version 

Looking for any possible lead for this issue.

zhongyuzeng, 2020-03-11:
Is this problem solved?I also encountered this problem

gonzalocasas, 2020-03-25:
This has been fixed and released in 1.0

closed 2020-03-25 by gonzalocasas