RobotWebTools/roslibpy #46: connecting to external server

melnarte, 2019-12-13:
Summary.
Error during publisher-listener between local (virtual machine with kinetic) and remote (server Digital Ocean with melodic) hosts 
## Expected Result
Successful communication publisher-listener between local and remote hosts

## Actual Result
artem@ubuntu16:~$ python talker.py 
Traceback (most recent call last):
  File "talker.py", line 7, in <module>
    client.run()
  File "/home/artem/.local/lib/python2.7/site-packages/roslibpy/ros.py", line 83, in run
    raise Exception('Failed to connect to ROS')
Exception: Failed to connect to ROS
## Reproduction Steps
(1) start rosbridge on Digital Ocean server and  start basic listener after
(2) start rosbridge on localhost
(3) run basic talker on localhost
```python
import time

import roslibpy

client = roslibpy.Ros(host="http://167.172.173.63", port=9090)
client.run()

talker = roslibpy.Topic(client, '/chatter', 'std_msgs/String')

while client.is_connected:
    talker.publish(roslibpy.Message({'data': 'Hello World!'}))
    print('Sending message...')
    time.sleep(1)

talker.unadvertise()

client.terminate()

```

## System Information
(localhost) virtual box with ubuntu 16 and ROS kinetic
(digital ocean host) ubuntu 18.04 with ROS melodic

gonzalocasas, 2019-12-14:
You need to remote the `http://` part of the `host` parameter, it's a websocket connection, so we use the host and port to build a URL like `ws://167.172.173.63:9090`, but in your case, it's resulting into `ws://http://167.172.173.63:9090` which is invalid.

gonzalocasas, 2019-12-16:
I'll close this, as it is most definitely a mistake in the `host` that was specified, however, feel free to keep commenting if you are still unable to get it to work.

closed 2019-12-16 by gonzalocasas