RobotWebTools/roslibpy #47: Guidance on Publishing messages with multiple componets

wegunterjr, 2020-01-10:
I am publishing data to a rosbridge, and seem to be just fine with 'std_msgs/String', and 'std_msgs/Float32, but when i try to do 'geometry_messages/Pose' which has orientation and position, I see the topic being published, but there is no data.

`       rvr_pose.position.x = locator_X
       rvr_pose.position.y = locator_Y
       rvr_pose.orientation.z = imu_Yaw
       print(locator_X)
       print(rvr_pose)
       print(rvr_pose.position)
       print(rvr_pose.orientation)
       print(type(rvr_pose.orientation))
    rvr_Pose.publish(roslibpy.Message({'position': rvr_pose.position, 'orientation' : rvr_pose.orientation}))
`
Thoughts?

gonzalocasas, 2020-01-10:
It's not clear what the type of `rvr_pose` is, but I assume it is not a dictionary, so very likely, the serialization is failing. Make sure you pass a dictionary in there and it will work.

wegunterjr, 2020-01-10:
Ah..It is Pose  from ROS, did I setup the Message right? With the two parts?

gonzalocasas, 2020-01-10:
ROS messages need to be converted to dictionaries to be sent via the `rosbridge`, the easiest way to do that is using the `yaml` module, so, in your case, you the following should work:

```
import yaml

rvr_pose = Pose()
rvr_pose.position.x = locator_X
rvr_pose.position.y = locator_Y
rvr_pose.orientation.z = imu_Yaw

rvr_pose_topic = roslibpy.Topic(client, '/rvr_pose', 'geometry_msgs/Pose')
rvr_pose_serialized = dict(yaml.load(str(rvr_pose)))
rvr_pose_topic.publish(roslibpy.Message(rvr_pose_serialized))
```

---

*NOTE: In your code, you use `rvr_pose` as a Pose instance, and `rvr_Pose` as a topic instance, that's not a very good idea, because they look almost the same, except the casing. In my example, I have renamed them to `rvr_pose` and `rvr_pose_topic` respectively.*

wegunterjr, 2020-01-11:
WOOHOO!!!! That worked!

gonzalocasas, 2020-01-11:
Great 👍

closed 2020-01-11 by gonzalocasas

danialdunson, 2022-07-14:
this should be added to the docs. thank you.