RobotWebTools/roslibpy #55: Support for python 2.7

paytience, 2020-03-28:
The README says roslibpy is supported for Python 2.7, however dependencies like txaio and autobahn only support Python 3.5+. Does this mean it is no longer supported?

## Expected Result

Get version number when using *roslibpy --version*, or be able to import roslibpy.

## Actual Result
casper@USERNAME-desktop:~/$ roslibpy --version
Traceback (most recent call last):
  File "/home/USERNAME/.local/bin/roslibpy", line 7, in <module>
    from roslibpy.__main__ import main
  File "/home/USERNAME/.local/lib/python2.7/site-packages/roslibpy/__init__.py", line 95, in <module>
    from .ros import Ros
  File "/home/USERNAME/.local/lib/python2.7/site-packages/roslibpy/ros.py", line 10, in <module>
    from .comm import RosBridgeClientFactory
  File "/home/USERNAME/.local/lib/python2.7/site-packages/roslibpy/comm/__init__.py", line 9, in <module>
    from .comm_autobahn import AutobahnRosBridgeClientFactory as RosBridgeClientFactory
  File "/home/USERNAME/.local/lib/python2.7/site-packages/roslibpy/comm/comm_autobahn.py", line 6, in <module>
    from autobahn.twisted.websocket import WebSocketClientFactory
  File "/home/USERNAME/.local/lib/python2.7/site-packages/autobahn/twisted/__init__.py", line 37, in <module>
    from autobahn.twisted.choosereactor import install_reactor
  File "/home/USERNAME/.local/lib/python2.7/site-packages/autobahn/twisted/choosereactor.py", line 30, in <module>
    import txaio
  File "/home/USERNAME/.local/lib/python2.7/site-packages/txaio/__init__.py", line 29, in <module>
    from txaio.interfaces import IFailedFuture, ILogger
  File "/home/USERNAME/.local/lib/python2.7/site-packages/txaio/interfaces.py", line 41, in <module>
    class IBatchedTimer(abc.ABC):
AttributeError: 'module' object has no attribute 'ABC'

## Reproduction Steps
```python
import roslibpy

```

## System Information
Python 2.7, Ubuntu 16.04, ROS Kinetic.

gonzalocasas, 2020-03-28:
Travis build is always built on python 2.7 as well so it most definitely work there. You need to install a version of twisted/autobahn that still supported python 2.7 of course, I believe the latest to support 2.x are versions 19.x.

gonzalocasas, 2020-03-28:
If you are having issues with the installation, try miniconda/anaconda installs, they might simplify the dep resolution.

paytience, 2020-03-29:
Okay thank you, I think the problem was the installation order, autobahn before txaio did the trick. Thanks for the help

closed 2020-03-29 by paytience