RobotWebTools/roslibpy #56: run, terminate, run again, it throws an exception is the right order

ramezanifar, 2020-04-02:
Hello. Thank you for the library.
Is it possible to run, terminate multiple times?
If I do the following in REPL
``` python
>> import roslibpy
>>> client = roslibpy.Ros(host='192.168.1.100', port=9090)
>>> client.run()  # As a result, client.is_connected is True
>>> client.terminate()  # As a result, client.is_connected is False
>>> client.run()
```

It throws an exception:

Exception in thread Thread-2:
Traceback (most recent call last):
  File "C:\Python27\lib\threading.py", line 801, in __bootstrap_inner
    self.run()
  File "C:\Python27\lib\threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "C:\Python27\lib\site-packages\twisted\internet\base.py", line 1282, in run
    self.startRunning(installSignalHandlers=installSignalHandlers)
  File "C:\Python27\lib\site-packages\twisted\internet\base.py", line 1262, in startRunning
    ReactorBase.startRunning(self)
  File "C:\Python27\lib\site-packages\twisted\internet\base.py", line 765, in startRunning
    raise error.ReactorNotRestartable()
ReactorNotRestartable

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Python27\lib\site-packages\roslibpy\ros.py", line 83, in run
    raise Exception('Failed to connect to ROS')
Exception: Failed to connect to ROS

## System Information
Operating = Windows 7 64
ROS version = Melodic
Python = 2.7

gonzalocasas, 2020-04-02:
This is by-design (of `Twisted`, not `roslibpy`), once a reactor is terminated, it cannot be restarted. But the you don't need to terminate the ros client, simply call `client.close()` and that closes the websocket connection and allows to reconnect.

closed 2020-04-02 by gonzalocasas

ramezanifar, 2020-04-02:
Thank you for the reply. I am going to depict some scenarios here. I would appreciate if you can comment.
scenario 1:
1- I start the connection
```python
>> import roslibpy
>>> client = roslibpy.Ros(host='192.168.1.100', port=9090)
>>> client.run()  # As a result, client.is_connected is True
```
2- I shutdown the rosbridge server in Linux. 
3- I close the connection in python.
```
>>> client.close()
```
4- I relaunch the rosbridge server in linux. 
After few seconds, it shows a client is connected and  "client.in_connected = True " event though I closed it. 
However, if I close while rosbridge is up and running, and then shutdown and relaunch, it behaves properly.

Thank you

ramezanifar, 2020-04-02:
scenario 2:
1- rosbridge server is down
2- I connect to it from python
```
>> import roslibpy
>>> client = roslibpy.Ros(host='192.168.1.100', port=9090)
>>> client.run()  # This times out and throws exception
```
It times out and throws exception because server is down.
3- I launch the rosbridge server in Linux.
4- After few seconds rosbridge server shows a client was connected. 
I think this is a problem because one may continue to attempt client.run() after exception.

gonzalocasas, 2020-04-07:
Scenario 1:
This actually is more or less what I would expect. On step 2, you shutdown `rosbridge` on linux side, so immediately, the ros client in python will notice and schedule a reconnection retry after some seconds. I assume that the call to `ros.close()` in this situation is not doing much since it's actively trying to reconnect already. Then, when you relaunch `rosbridge` it reconnect as expected.

Scenario 2:
Again, this is exactly as intended. You told the client to connect, but the `rosbridge` is down, so it will schedule reconnects with an exponential back-off. Consequently, when you launch `rosbridge`, the client connects after some seconds. 

If you're interested in the inner workings of the library, consider enabling `DEBUG` logging to better understand what is going on. For example:

```python
import roslibpy
import logging

log = logging.getLogger('roslibpy')
log.setLevel(logging.DEBUG)
ch = logging.StreamHandler()
log.addHandler(ch)

client = roslibpy.Ros(host='127.0.0.1', port=9090)
client.run()
```

ramezanifar, 2020-04-07:
Thank you for the reply.
**Scenario 1:**
This behavior is problematic. The ros.close() is not doing what it is supposed to do.  I force it to close and it reconnects anyways. 

**Scenario 2:**
This behavior is also problematic. It is not known how long I should wait. 10 seconds, 30 second, indefinitely. If I attempt to reconnect while your auto reconnect is in the background and it succeeds, it may end up creating two connected clients in the server. Whereas, if I know connection is down, I can retry to connect immediately.

**Summary**
I know many people have requested the reconnect feature. It helps many use cases but also hurts in a case where more control is required. 

**Suggestion**
That would be nice if the reconnect feature is an optional setting. 
I feel more powerful with your library if I can control the connection as I need it. Something as simple as: 
```python
While running == True
   if ros.is_connected == False:
      ros.connect()
   else:
     # do your tasks
```
would be sufficient in most applications.
I appreciate your good work.