RobotWebTools/roslibpy #62: ros.close() does not disconnect the instance from rosbrige's websocket

gcgloven, 2020-07-31:
Summary.

## Expected Result
expect is_connected to show `False` after close() the connection

## Actual Result
What happened instead.
is_connected remain showing `True`
Cannot disconnect from ros_websocket.. hence, my newly created ros object for the same address will fail to connect 

## Reproduction Steps
```python
import roslibpy
class RosBridge(object):
    def __init__(self,websocket_ip,port):
        self.ros_address= 'ws://' + str(websocket_ip) + ':' + str(port)
        self.ros = roslibpy.Ros(host=websocket_ip, port=port)
        self.status =  None
        
    def connect(self):
        print(f"trying to connect to ROS {str(self.ros_address)}")
        self.ros.run()
        self.status =  self.isConnected()
    def disconnect(self):
        print(f"trying to disconnect from to ROS {str(self.ros_address)}")
        self.ros.close()
        self.ros.terminate()
        self.status =  self.isConnected()
    def isConnected(self):
        """
        Check connectivity of ROS bridge.
        """
        status= self.ros.is_connected 
        if status == True:
            print("ROS Bridge is connected")
        else:
            print("ROS Connectivity Failed")
        return status
  
if __name__ == '__main__':    
     robot = RosBridge(websocket_ip="localhost",port=9090)
     robot.connect()
     robot.disconnect()
```
## System Information
Operating System name and version, ROS version, etc.
ROS Kinetics default 
```roslaunch rosbridge_server rosbridge_websocket.launch```

beverlylytle, 2020-07-31:
Hi! Thanks for using RosLibPy!  The call `Ros.close` is asynchronous, and it can take some time between invoking `close` and `is_connected` returning `False`.  Try waiting until `is_connected` returns `False` before connecting a new object to the same address.  Please let me know if this has helped.

gcgloven, 2020-08-11:
> Hi! Thanks for using RosLibPy! The call `Ros.close` is asynchronous, and it can take some time between invoking `close` and `is_connected` returning `False`. Try waiting until `is_connected` returns `False` before connecting a new object to the same address. Please let me know if this has helped.
Thank you so much for the fast reply! I have implemented a while true loop to check this (not a good design), it is now working to off a connection.

beverlylytle, 2020-08-17:
Good to hear!  Since things seem to be working for you, I will close this issue.

closed 2020-08-17 by beverlylytle