RobotWebTools/roslibpy #63: Double JSON-Encoded strings.

travipross, 2020-08-13:
When using the `set()` method of `roslibpy.ros.core.Param()`, it seems that string parameters on the other side of the `rosbridge_server` receive the value as a double JSON-encoded string. I'm wondering why the `ServiceRequest` is instantiated with the value `json.dumps`ed?
```request = ServiceRequest(
            {'name': self.name, 'value': json.dumps(value)})
```
https://github.com/gramaziokohler/roslibpy/blob/909b87fdcd1ad42e23d14f2ddb83b1ab90463d3f/src/roslibpy/core.py#L407

For an example of what I'm getting:
```>>> import roslibpy
>>> client = roslibpy.Ros(host='localhost', port=9090)
>>> client.run()
>>> client.is_connected
True
>>> client.get_param("/test_node:host")
'localhost'
>>> client.set_param("/test_node:host", "localhost")
>>> client.get_param("/test_node:host")
'"localhost"'
```

This is being run alongside the `ros2` branch of `rosbridge_suite` [(Link)](https://github.com/RobotWebTools/rosbridge_suite/tree/ros2).

If this isn't an issue with `roslibpy`, I'm wondering if anyone might have some suggested troubleshooting steps?

travipross, 2020-08-19:
This appears to be a problem in `rosbridge_suite` that was introduced when refactoring to use ROS2. I'm not sure if anything needs to be fixed on the `roslibpy` side, or just on `rosbridge_suite`.

Looking at the `git blame` for the portion of `rosbridge_suite` that handles these parameter requests, you can see that some additional logic was added to guess the type of payload being sent. I think there was just an oversight in the implementation of this which incorrectly parsed JSON-encoded string types, and returned them un-parsed.

I've opened a PR (https://github.com/RobotWebTools/rosbridge_suite/pull/521) which solved the issue for me, though some tests are failling on that project's CI.

gonzalocasas, 2020-08-19:
@travipross thanks for tracking it down and sending that PR to `rosbridge_suite`!
We can keep this open until your PR is merged, and then it should be ok to close.

travipross, 2020-08-26:
Fix has been implemented in `rosbridge_suite`, so this seems to be resolved. [Link](https://github.com/RobotWebTools/rosbridge_suite/pull/521)

closed 2020-08-26 by travipross