RobotWebTools/roslibpy #67: reset reconnect retry timer

chia-labradorsystems, 2020-11-30:
I notice that in roslibpy

> Reconnect will be retried with an exponential back-off.

Is there any way that can rest the timer without modifying the code?

gonzalocasas, 2020-11-30:
There are three methods that can be used to modify the retry behavior/timing: `set_max_delay`, `set_initial_delay`, and `set_max_retries`. They belong to the factory, so, you can use them like this: 
```
ros = roslibpy.Ros('127.0.0.1', 9090)
ros.factory.set_initial_delay(5)  # 5 seconds init delay
```

gonzalocasas, 2020-11-30:
Let us know if that works for you, and if it does, we close the issue.

chia-labradorsystems, 2020-11-30:
Thanks, let me test it :)

chia-labradorsystems, 2020-11-30:
yes I think it works
I just notice that this is added since 1.1.0, and I was using 1.0.0

closed 2020-11-30 by chia-labradorsystems

gonzalocasas, 2020-11-30:
great! thanks!