RobotWebTools/roslibpy #73: Is it possible to make robot move a certain distance like rospy does?

PatZhuang, 2021-01-12:
I tried to use loop like this, which works when using rospy:

```python
cmd_topic = roslibpy.Topic(client, '/cmd_vel', 'geometry_msgs/Twist', throttle_rate=50)
cmd_topic.advertise()

dis = 0.1
linear_speed = 0.1
msg = roslibpy.Message({
    'linear':{'x': linear_speed, 'y': 0.0, 'z': 0.0},
    'angular': {'x': 0.0, 'y': 0.0, 'z': 0.0}
})
linear_duration = dis / linear_speed
# move forward
ticks = int(cmd_topic.throttle_rate * linear_duration)
for i in range(ticks):
    self.cmd_topic.publish(msg)
    time.sleep(cmd_topic.throttle_rate / 1000)
msg['linear']['x'] = 0.0
cmd_topic.publish(msg)
time.sleep(1)
```

However the robot moves way more than 0.1 meter but about 0.3 meter. Am I doing wrong or roslibpy does not support operation like this?

beverlylytle, 2021-01-12:
Hi!  Thanks for making an issue in `roslibpy`.  We are always happy to have feedback! I believe it should be possible to send the type of messages you've defined.  However I think you've miscalculated the number of `ticks`.  I believe it should
```
ticks = int((cmd_topic.throttle_rate / 1000) * linear_duration)
```
With your definition of `ticks`, you send the message 50 times, but with the above definition it would be sent 20 times, which  would account for the approximate overshooting the mark by about .15 m.  Does this make sense to you? Or have I also missed the mark? ;)

PatZhuang, 2021-01-13:
> Hi! Thanks for making an issue in `roslibpy`. We are always happy to have feedback! I believe it should be possible to send the type of messages you've defined. However I think you've miscalculated the number of `ticks`. I believe it should
> 
> ```
> ticks = int(linear_duration / (cmd_topic.throttle_rate / 1000) )
> ```
> 
> With your definition of `ticks`, you send the message 50 times, but with the above definition it would be sent 20 times, which would account for the approximate overshooting the mark by about .15 m. Does this make sense to you? Or have I also missed the mark? ;)

Oops.....You're right, I mis-calculated ticks. I changed the formulation and everything works fine. Thank you very much.

closed 2021-01-13 by PatZhuang