RobotWebTools/roslibpy #77: module 'roslibpy' has no attribute 'Header'

pratipo, 2021-02-10:
I cannot generate a properly built message, apprently

Steps to reproduce the behavior:
I have a simulated robot arm in gazebo
`rostopic pub /arm_controller/command trajectory_msgs/JointTrajectory '{joint_names: ["joint_1","joint_2","jnt_3","joint_4", "joint_5", "joint_6"], points: [{positions:[0.1,-0.5,0.5,0.75,0.0,0.0], time_from_start: [1.0,0.0]}]}' -1` works as expected

i got some code from `/tests/test_topic.py`

`msg = dict(#header=roslibpy.Header(seq=i, stamp=Time.now(), frame_id='base'),
                       joint_names=["joint_1","joint_2","joint_3","joint_4", "joint_5", "joint_6"],
                       points=[dict(
                                    positions=[1.0,0.0,2.0,0.0,1.0,1.0],
                                    # velocities=[1,1,1,1,1,1], 
                                    # accelerations=[],
                                    # effort=[],
                                    time_from_start=[1.0,0.0])],
                       )
            publisher.publish(roslibpy.Message(msg))
            time.sleep(1)`

i had to remove the header, because i get
`module 'roslibpy' has no attribute 'Header'`

`roslibpy.core.Header` doesnt work either
`
rostopic echo /arm_controller/command` displays the messages, but gazebo is not moving the arm

any ideas?
thanks a lot in advance

 - OS: `linux mint 20`
 - Python version `3.7.4`
 - Python package manager `pip3`

gonzalocasas, 2021-02-10:
oh! The header class was just recently implemented and it's not been released (it would work if you run from source). But we're ready to release, so, give me a few minutes and we'll push a new release out.

gonzalocasas, 2021-02-11:
@pratipo should be fixed now. Please update to the latest release: `roslibpy 1.2.0` to get these feature.

closed 2021-02-11 by gonzalocasas

pratipo, 2021-02-11:
just to confirm it's working. thanks!