RobotWebTools/roslibpy #78: Implementation of movebase ros node to send nav goals

KSaikishore, 2021-02-15:
How can i send nav goals using roslibpy. I have looked at the below mentioned tutorials but they were mainly relying on **move_base_msgs** package which isn't available on roslibpy. Is there a way or example on how to send goals to navigation stack using roslibpy.

1. http://wiki.ros.org/navigation/Tutorials/SendingSimpleGoals
2. https://hotblackrobotics.github.io/en/blog/2018/01/29/action-client-py/

gonzalocasas, 2021-02-15:
@KSaikishore once `move_base` is installed on your ROS setup, it will be available on `roslibpy`, there's not extra support needed, you just need to call the services, actions, and topics provided by the package and pass a python dictionary that respects the exact same fields as defined in the package and that should work.

gonzalocasas, 2021-02-17:
I'm closing this issue, because I believe this works already, but please do feel free to keep commenting here if you're facing trouble when trying it out.

closed 2021-02-17 by gonzalocasas