RobotWebTools/roslibpy #81: no data published on topic when trying to re-publish scan topic data.

kjitender469, 2021-04-05:
Summary.
I am trying to read scan data from /scan topic and publish it onto /fake_scan topic (i.e. redirect scan topic data to fake_scan). But data is not being published at all. I received ranges data from scan into local list and the size of range data is 300. What I did is I sliced the list to 188 elements and try to publish and this time data successfully published to /fake_scan but if I increase the sliced data from 188 to 189 the data is not published. I don't know why this is happening. Is there any limitation of data size that has to be published using roslibpy? 

## System Information
Operating System name and version : ubuntu16.04
ROS version : kinetic

gonzalocasas, 2021-04-05:
What's the data type of the topic? And what's the unit of your measures? 188 Mb? 188 points?

kjitender469, 2021-04-05:
data type of the topic is sensor_msgs/LaserScan. 188 points.

gonzalocasas, 2021-04-06:
@kjitender469 could you record a ros bag with the 300+ points to try to reproduce this? I don't think there's any limit, at least, not a deliberate one anywhere in the pipeline, but I can take a look.

kjitender469, 2021-04-06:
[Client 0] [id: publish:/fake_scan:381] publish: sensor_msgs/LaserScan message requires a float32 for field ranges, but got a <type 'NoneType'>

This is the message that displays on the terminal where rosbridge webscoket server is running.

gonzalocasas, 2021-04-06:
ah ok! then it's not related to limits, there's a data issue. Please check why that expected float32 is none; it cannot be a problem on the source topic because ROS itself would have rejected it before; maybe something in the slicing process is leaving some nulls scattered in the data?

kjitender469, 2021-04-06:
I have figure out the issue. Actually range list itself have some None values so when I sliced the range list these None values can also be coppied into my sliced list. Initially upto 189 point there are no None value. None value are returned from where there is no obstacle in the laser path for example open space. Thats why None values are also present in the ranges data.
Thank you @gonzalocasas for helping me out.

closed 2021-04-06 by kjitender469

gonzalocasas, 2021-04-06:
glad to hear it works! 👍