RobotWebTools/roslibpy #89: Sending a list in ServiceRequest

philianeles, 2022-02-04:
Hi, I am trying to send a list of "va" and "ha" values in a `ServiceREquest`, on line 51 below:

[image]

And running into the error that can also seen above.
Is there a way to send a list in a `ServiceRequest`?

Thank you for the help!``

philianeles, 2022-02-04:
Yes! my bad, the correct version should have been this:
```
service = roslibpy.Service(ros_client, '/plt_ros/cmd/pose_set_absolute', 'plt_msgs/SetPose')
for va, ha in zip(VA_user_input, HA_user_input):
      request = roslibpy.ServiceRequest(dict(VA=va, HA=ha))
      call_service(service, request)
```
which would imply having keys that change dynamically based on the content of VA_user_input afterall...
`plt_msgs/SetPose` is a (custom) service msg:
```
float32 HA
float32 VA
---
ResponseCode response
```

gonzalocasas, 2022-02-04:
aha! so, does that change fix the problem? can we close the issue? or is there something else still open?

philianeles, 2022-02-04:
not sure yet - I could not test it with the hardware yet, not connected atm, but will follow-up here asap!

closed 2022-02-08 by philianeles