RobotWebTools/roslibpy #90: secs not incrementing: roslibpy.Time(time.to_sec())

philianeles, 2022-02-09:
Hi, I am trying to increment `secs` in the message below, which seems to be not changing at all:
```
    header:
      seq: 90
      stamp:
        secs: 1644429708
        nsecs: 238979101
      frame_id: "ur_tool0"
    pose:
      position:
        x: 0.374179967324
        y: -0.892998350452
        z: 0.786607771367
      orientation:
        x: 0.5
        y: -0.5
        z: 0.5
        w: 0.5
  -
    header:
      seq: 91
      stamp:
        secs: 1644429708
        nsecs: 239021062
      frame_id: "ur_tool0"
    pose:
      position:
        x: 0.407864024509
        y: -0.892998350452
        z: 0.786607771367
      orientation:
        x: 0.5
        y: -0.5
        z: 0.5
        w: 0.5
  -
    header:
      seq: 92
      stamp:
        secs: 1644429708
        nsecs: 239062070
      frame_id: "ur_tool0"
    pose:
      position:
        x: 0.441548081695
        y: -0.892998350452
        z: 0.786607771367
      orientation:
        x: 0.5
        y: -0.5
        z: 0.5
        w: 0.5
  -
    header:
      seq: 93
      stamp:
        secs: 1644429708
        nsecs: 239103078
      frame_id: "ur_tool0"
    pose:
      position:
        x: 0.47523213888
        y: -0.892998350452
        z: 0.786607771367
      orientation:
        x: 0.5
        y: -0.5
        z: 0.5
        w: 0.5
  -
    header:
      seq: 94
      stamp:
        secs: 1644429708
        nsecs: 239144086
      frame_id: "ur_tool0"
    pose:
      position:
        x: 0.508916196065
        y: -0.892998350452
        z: 0.786607771367
      orientation:
        x: 0.5
        y: -0.5
        z: 0.5
        w: 0.5
  -
    header:
      seq: 95
      stamp:
        secs: 1644429708
        nsecs: 239185094
      frame_id: "ur_tool0"
    pose:
      position: 
        x: 0.542600253251
        y: -0.892998350452
        z: 0.786607771367
      orientation:
        x: 0.5
        y: -0.5
        z: 0.5
        w: 0.5
  -
    header:
      seq: 96
      stamp:
        secs: 1644429708
        nsecs: 239226102
      frame_id: "ur_tool0"
    pose:
      position:
        x: 0.576284310436
        y: -0.892998350452
        z: 0.786607771367
      orientation:
        x: 0.5
        y: -0.5
        z: 0.5
        w: 0.5
  -
    header:
      seq: 97
      stamp:
        secs: 1644429708
        nsecs: 239267110
      frame_id: "ur_tool0"
    pose:
      position:
        x: 0.609968367621
        y: -0.892998350452
        z: 0.786607771367
      orientation:
        x: 0.5
        y: -0.5
        z: 0.5
        w: 0.5
  -
    header:
      seq: 98
      stamp:
        secs: 1644429708
        nsecs: 239308118
      frame_id: "ur_tool0"
    pose:
      position:
        x: 0.643652424806
        y: -0.892998350452
        z: 0.786607771367
      orientation:
        x: 0.5
        y: -0.5
        z: 0.5
        w: 0.5
  -
    header:
      seq: 99
      frame_id: "ur_tool0"
    pose:
      position:
        x: 0.677336481992
        y: -0.892998350452
        z: 0.786607771367
      orientation:
        x: 0.5
        y: -0.5
        z: 0.5
        w: 0.5
  -
    header:
      seq: 100
      stamp:
        secs: 1644429708
        nsecs: 239392995
      frame_id: "ur_tool0"
    pose:
      position:
        x: 0.711020539177
        y: -0.892998350452
        z: 0.786607771367
      orientation:
        x: 0.5
        y: -0.5
        z: 0.5
        w: 0.5
```

The code that publishes the message is like this:
```
                    publisher = Topic(ros_client, '/mission_control/planned_path', 'nav_msgs/Path')
                    pose_array = []
                    
                    time = roslibpy.Time(1, 0.1).now()
                    time = roslibpy.Time(time.to_sec() + 5, 0)
                    seq = 0
                    for i in pos_list:
                        if i < start_seq: 
                            pose_array.append({'header': {'seq': seq, 'stamp': time.to_sec(),'frame_id': 'ur_tool0'}, 'pose': {'position': {'x': pos_x[i], 'y': pos_y[i], 'z': pos_z[i]}, 'orientation': {'x': trajectory_orientation[i][1], 'y': trajectory_orientation[i][2], 'z': trajectory_orientation[i][3], 'w': trajectory_orientation[i][0]}}})
                        elif i >= start_seq and i <= stop_seq:
                            pose_array.append({'header': {'seq': seq, 'stamp': time.to_sec(),'frame_id': 'ur_tool0'}, 'pose': {'position': {'x': pos_x[i], 'y': pos_y[i], 'z': pos_z[i]}, 'orientation': {'x': trajectory_orientation[i][1], 'y': trajectory_orientation[i][2], 'z': trajectory_orientation[i][3], 'w': trajectory_orientation[i][0]}}})
                            #start spraying
                            """
                            service = Service(ros_client, '/ur_hardware_interface/set_io', 'ur_msgs/SetIO')
                            request = ServiceRequest(dict(fun = 1, pin = 0, state = 0))
                            call_service(service, request)
                            service = Service(ros_client, '/ur_hardware_interface/set_io', 'ur_msgs/SetIO')
                            request = ServiceRequest(dict(fun = 1, pin = 1, state = 1))
                            call_service(service, request)
                            service = Service(ros_client, '/ur_hardware_interface/set_io', 'ur_msgs/SetIO')
                            request = ServiceRequest(dict(fun = 1, pin = 4, state = 1))
                            call_service(service, request)
                            """
                        elif i > stop_seq:
                            pose_array.append({'header': {'seq': seq, 'stamp': time.to_sec(),'frame_id': 'ur_tool0'}, 'pose': {'position': {'x': pos_x[i], 'y': pos_y[i], 'z': pos_z[i]}, 'orientation': {'x': trajectory_orientation[i][1], 'y': trajectory_orientation[i][2], 'z': trajectory_orientation[i][3], 'w': trajectory_orientation[i][0]}}})
                            #stop spraying
                            """
                            service = Service(ros_client, '/ur_hardware_interface/set_io', 'ur_msgs/SetIO')
                            request = ServiceRequest(dict(fun = 1, pin = 0, state = 1))
                            call_service(service, request)
                            service = Service(ros_client, '/ur_hardware_interface/set_io', 'ur_msgs/SetIO')
                            request = ServiceRequest(dict(fun = 1, pin = 1, state = 0))
                            call_service(service, request)
                            service = Service(ros_client, '/ur_hardware_interface/set_io', 'ur_msgs/SetIO')
                            request = ServiceRequest(dict(fun = 1, pin = 4, state = 0))
                            call_service(service, request)
                            """
                        time = roslibpy.Time(0, 0).from_sec(time.to_sec() + 1)
                        seq += 1
                    message= publisher.publish({'header': {'seq': seq, 'stamp': time.to_nsec() ,'frame_id': 'odom'}, 'poses': pose_array})

```

gonzalocasas, 2022-02-10:
I think there's a misunderstanding with the values expected by the `Time()` default constructor. It expects two integers, and there are several lines in your example that use floats, which I guess end up in some rather invalid state of the class. We should add validation to raise a clearer error if this happens thou, but in the meantime, the following lines should be double checked:

- `time = roslibpy.Time(1, 0.1).now()`:  this works, but it is not correct. `now` should be called as a static method: `roslibpy.Time.now()`
- `time = roslibpy.Time(time.to_sec() + 5, 0)`: I guess this is meant to be `time = roslibpy.Time.from_sec(time.to_sec() + 5)`
- `time = roslibpy.Time(0, 0).from_sec(time.to_sec() + 1)`: this one works, but can be shorter: `time = roslibpy.Time.from_sec(time.to_sec() + 1)`

and then, when you're creating the messages, why not use the class `Header` and `Time` there as well? So instead of manually turning them into dicts, which I guess leaves room for mistakes (e.g. when you append to the `pose_array`, the value of `stamp` is set to `'stamp': time.to_sec()` which I doubt converts correctly to a time type, you can do something like: 
```python
      time = roslibpy.Time.from_sec(time.to_sec() + 1)
      seq += 1
      message = publisher.publish({'header': roslibpy.Header(seq=seq, stamp=time, frame_id='odom'), 'poses': pose_array})
```
and same thing goes for the lines with `pose_array.append(`

philianeles, 2022-02-11:
Thank you @gonzalocasas! all these seem to fix the problem!

closed 2022-02-11 by philianeles