RobotWebTools/roslibpy #94: call this funtion client.terminate()  or close(),  then can not reconnect.

liuxinxiao, 2022-06-08:
My code :
client = roslibpy.Ros(host='localhost', port=9090)
client.run()
print('Is ROS connected?', client.is_connected)
client.terminate()

#reconnect ros:
client = roslibpy.Ros(host='localhost', port=9090)
client.run()
print('Is ROS connected?', client.is_connected)

client.is_connecting is True,  client.is_connected is False.

gonzalocasas, 2022-06-08:
@liuxinxiao it depends on your context (ie. IronPython vs CPython), but in the normal case, calling `terminate()` is a final operation that signals the termination of the main event loop. In other words, you're really saying "My code is going to finish completely" and there's no reconnection possibility (see [docs](https://roslibpy.readthedocs.io/en/latest/reference/index.html#roslibpy.Ros.terminate)). So, in this case, the behavior you are seeing is correct. 

Now, you also mention this happens if you use `.close()`. In that case, reconnections should work. Could you please confirm that using `.close()` only without `terminate()` also causes it to fail to reconnect?

liuxinxiao, 2022-06-08:
**the first connect. It work.**
2022-06-08 20:39:57,383 autoware_adapter.py[line:17] ERROR ros_client is None IP : XXX
2022-06-08 20:39:57,383 autoware_adapter.py[line:18] ERROR ros_client is None port is : 9090
2022-06-08 20:39:57,402 comm_autobahn.py[line:30] INFO Connection to ROS ready.
2022-06-08 20:39:57,403 autoware_adapter.py[line:22] ERROR Is ROS connected ? True

stop function.  I use ros.close(),  But is_connected is True still.
2022-06-08 20:41:18,788 autoware_adapter.py[line:82] ERROR Stop function Is ROS connected ? True
2022-06-08 20:41:18,789 autoware_adapter.py[line:84] ERROR Stop function close Is ROS connected ? True
2022-06-08 20:41:18,797 comm_autobahn.py[line:45] INFO WebSocket connection closed: Code=1000, Reason=None
2022-06-08 20:41:18,798 _legacy.py[line:147] INFO Stopping factory <roslibpy.comm.comm_autobahn.AutobahnRosBridgeClientFactory object at 0x7f1767f83b50>
2022-06-08 20:41:18,883 server.py[line:259] INFO connection closed
2022-06-08 20:42:43,154 server.py[line:641] INFO connection open

**The second connect.   log is :**
2022-06-08 20:42:46,178 autoware_adapter.py[line:17] ERROR ros_client is None IP : XXX
2022-06-08 20:42:46,178 autoware_adapter.py[line:18] ERROR ros_client is None port is : 9090
2022-06-08 20:42:46,180 _legacy.py[line:147] INFO Starting factory <roslibpy.comm.comm_autobahn.AutobahnRosBridgeClientFactory object at 0x7f174874c2e0>

**The second connect,  Throw except.**
/usr/local/lib/python3.8/dist-packages/pymongo/compression_support.py:56: UserWarning: Unsupported compressor: disabled
  warnings.warn("Unsupported compressor: %s" % (compressor,))
/usr/local/lib/python3.8/dist-packages/pymongo/common.py:803: UserWarning: Unknown option gssapiservicename
  warnings.warn(str(exc))
Exception in thread Thread-94:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/sdg_engine_code/Engine/src/main/engine.py", line 86, in run
    self.ads_adapter = AutowareAdapter(ros_bridge_ip, log=self.log, task_id=self.task_id)
  File "/sdg_engine_code/Engine/src/main/autoware_adapter.py", line 21, in __init__
    self.ros_client.run()
  File "/usr/local/lib/python3.8/dist-packages/roslibpy/ros.py", line 91, in run
    raise Exception('Failed to connect to ROS')
Exception: Failed to connect to ROS

**My code is :** 
def __init__(self, ip, ws_port=9091, ros_port=9090, log=None, task_id=None):
	self.ros_client = None
	logging.error("ros_client is None IP : {}".format(ip))
	logging.error("ros_client is None port is : {}".format(ros_port))

	self.ros_client = roslibpy.Ros(host=ip, port=ros_port)
	self.ros_client.run()
	logging.error('Is ROS connected ? {}'.format(self.ros_client.is_connected))
		
def stop(self):

        try:
            # self.trace_listener.unsubscribe()
            self.state_listener.unsubscribe()
            logging.error('Stop function Is ROS connected ? {}'.format(self.ros_client.is_connected))
            self.ros_client.close()
            logging.error('Stop function close Is ROS connected ? {}'.format(self.ros_client.is_connected))
        except Exception as exception:
            print(exception)
            self.log.error_log({"msg": "{}".format(traceback.format_exc()), "task_id": self.task_id})

gonzalocasas, 2022-06-08:
@liuxinxiao could you please paste the entire code? I don't totally get what is going on without more context.

falfab, 2025-03-06:
I would like to follow up on this issue since we are facing the same problem.

I cannot use `close()` or `terminate()` to close and/or re-open a connection to rosbridge, and in general seems that there are some issues regarding the connection/disconnection logic.

## `is_connected` does not get updated

Sample code:

```py
import roslibpy

client = roslibpy.ros.Ros("localhost", 9090)
client.run()
client.close()
print(f"{client.is_connected}")
```

Expected output: `false`

Actual output: `true`

## Cannot reconnect after connection closed

Sample code:

```py
import roslibpy

client = roslibpy.ros.Ros("localhost", 9090)
client.run()
client.close()
client.connect() 
```

Expected: client reconnects

Actual: client does not receonnect

The issue here is related to the former one, since `is_connected` does not get updated during closing, then `connect()` does nothing.

falfab, 2025-03-06:
I would like to follow up to my comment a few moments ago. 

After further investigation I found out that `close` and `connect` are async.

Adding a 

```
while client.is_connected:
```

solved the issue.