RobotWebTools/roslibpy #96: Cannot use external action

DaAwesomeP, 2022-06-15:
**Description**

I am trying to use an action from `irobot_create_msgs`, but I am not quite sure how I am supposed to format `ActionClient()` to properly associate the action with `irobot_create_msgs`. If this is instead a bug with rosbridge, please let me know.

**To Reproduce**
Steps to reproduce the behavior:
1. Create venv, install `roslibpy`
2. Create file `ros.py`:
```python3
from __future__ import print_function
import roslibpy
import roslibpy.actionlib

client = roslibpy.Ros(host='127.0.0.1', port=9090)
client.run()

action_client = roslibpy.actionlib.ActionClient(client,
                                                '/myrobot',
                                                'irobot_create_msgs/DriveDistanceAction')

goal = roslibpy.actionlib.Goal(action_client,
                               roslibpy.Message({'distance': 0.5}))

goal.on('feedback', lambda f: print(f['remaining_travel_distance']))
goal.send()
result = goal.wait(10)
action_client.dispose()

print('Result: {}'.format(result['pose']))
```
3. Errors when running file:
```bash
$ python3 ros.py
Traceback (most recent call last):
  File "/home/pi/repo/ros.py", line 8, in <module>
    action_client = roslibpy.actionlib.ActionClient(client,
  File "/home/pi/venv/lib/python3.9/site-packages/roslibpy/actionlib.py", line 234, in __init__
    raise Exception('Action client failed to connect, no status received.')
Exception: Action client failed to connect, no status received.
```

**Expected behavior**

**Screenshots**

rosbridge log:
```
docker-rosbridge-1  | [rosbridge_websocket-1] [INFO] [1655330361.880834837] [rosbridge_websocket]: Client connected. 1 clients total.
docker-rosbridge-1  | [rosbridge_websocket-1] [ERROR] [1655330361.888610688] [rosbridge_websocket]: [Client 74d4ac77-fb09-4970-845a-ccd9ebcf8f7f] [id: advertise:/myrobot/goal:1] advertise: Unable to import msg class DriveDistanceActionGoal from package irobot_create_msgs. Caused by module 'irobot_create_msgs.msg' has no attribute 'DriveDistanceActionGoal'
docker-rosbridge-1  | [rosbridge_websocket-1] [INFO] [1655330361.899699162] [rosbridge_websocket]: [Client 74d4ac77-fb09-4970-845a-ccd9ebcf8f7f] Subscribed to /myrobot/status
docker-rosbridge-1  | [rosbridge_websocket-1] [ERROR] [1655330361.902967490] [rosbridge_websocket]: [Client 74d4ac77-fb09-4970-845a-ccd9ebcf8f7f] [id: subscribe:/myrobot/feedback:4] subscribe: Unable to import msg class DriveDistanceActionFeedback from package irobot_create_msgs. Caused by module 'irobot_create_msgs.msg' has no attribute 'DriveDistanceActionFeedback'
docker-rosbridge-1  | [rosbridge_websocket-1] [ERROR] [1655330361.906295577] [rosbridge_websocket]: [Client 74d4ac77-fb09-4970-845a-ccd9ebcf8f7f] [id: subscribe:/myrobot/result:5] subscribe: Unable to import msg class DriveDistanceActionResult from package irobot_create_msgs. Caused by module 'irobot_create_msgs.msg' has no attribute 'DriveDistanceActionResult'
docker-rosbridge-1  | [rosbridge_websocket-1] [INFO] [1655330364.476330826] [rosbridge_websocket]: Client disconnected. 0 clients total.
```

**System (please complete the following information):**
 - OS: Raspberry Pi OS (Debian) Lite aarch64 (amd64)
 - Python version: 3.9.2
 - Python package manager: pip

**Additional context**

rosbridge is running in a Docker container on the same machine with networking in host mode and the following Dockerfile:
```Dockerfile
FROM ros:galactic-ros-core

RUN apt-get update && apt-get install -y \
      ros-${ROS_DISTRO}-rosbridge-suite \
      ros-${ROS_DISTRO}-irobot-create-msgs && \
    rm -rf /var/lib/apt/lists/*

EXPOSE 9090
EXPOSE 11311

CMD ["ros2", "launch", "rosbridge_server", "rosbridge_websocket_launch.xml"]
```
The bridge is definitely functioning and connecting with the iRobot, as I see the iRobot topics in the correct namespace when I run `roslibpy topic list`. I also verified that the action is installed:

```bash
$ cat /opt/ros/galactic/share/irobot_create_msgs/action/DriveDistance.action
# Request

# Drive a distance (meters) in a straight line
float32 distance
# Max translation speed (positive m/s), will cap negative distance to negative speed
float32 max_translation_speed 0.3
---
# Result
# Pose where robot finished
geometry_msgs/PoseStamped pose
---
# Feedback
# How much distance is left to travel
float32 remaining_travel_distance
```

DaAwesomeP, 2022-06-27:
Appears to be related to https://github.com/RobotWebTools/rosbridge_suite/issues/552 and https://github.com/RobotWebTools/rosbridge_suite/issues/697

mhubii, 2023-05-23:
might be related to https://github.com/RobotWebTools/rosbridge_suite/pull/804

ender18g, 2025-01-22:
@DaAwesomeP any luck with this issue? We are having the exact same problem with create3s running Humble, and rosbridge suite running in a docker container just like yours.

DaAwesomeP, 2025-01-22:
@ender18g Unfortunately I haven't used ROS in many years. I don't think that we solved the issue.