RobotWebTools/roslibpy #98: I had the same problem and I was sure the address was righ

xibei312, 2022-07-11:
**Description**
connect() doesn't reconnect 

**To Reproduce**
Steps to reproduce the behavior:
ros.close()
// Connection is closed
// Wait for ~2 seconds
ros.connect()
// Reconnected to websocket

**Expected behavior**
ros.close()
// Connection is closed
// Wait for ~2 seconds
ros.connect()
// Doesn't reconnect
ros.is_connected outputs False

**Reproduction Steps**
import roslibpy
import time
ros = roslibpy.Ros(host, port)
ros.run()
ros.close()
time.sleep(5)
ros.connect()

**System (please complete the following information):**
 - OS: `[ubuntu20.04]`
 - Python version `[3.8]`
 - Python package manager `[pip]`
 - roslibpy version`[1.3.0]`

**Additional context**

dkarakay, 2022-08-19:
I have the same problem

dkarakay, 2022-08-22:
Are you going to publish a new update? @gonzalocasas

gonzalocasas, 2022-08-23:
@dkarakay I cannot reproduce the problem. I just executed the following script on the python interpreter, which is basically the exact same as the "Reproduction steps" posted by @xibei312, but it reconnects correctly as far as I can see:

[image]

Also rosbridge reports correctly the connection/disconnection:
[image]

I think the only problem is that you are not waiting enough time to check if it's really re-connected, because if I add a few `time.sleep` statements in the reproduction steps and run as script, it also behaves correctly:

[image]

@dkarakay @xibei312 do you agree with this? is it ok to close the issue?

dkarakay, 2022-08-23:
No, most of the time waiting is enough, but at some point, the client enters a loop in which it stucks in the connecting process and cannot not establish the connection in the end. It has mostly happened when we have multiple clients etc. What I mean is that the system is not stable.

gonzalocasas, 2022-08-23:
@dkarakay would you be able to provide a script that reproduces this behavior?

Because I haven't really seen that unstable behavior so far. Not on the autobahn-based implementation of the connection at least (I did noticed some issues on the .net implementation of reconnections). If I understand correctly, you are running this from CPython (not Ironpython), so you should be using the autobahn part.

dkarakay, 2022-08-24:
Actually, our system is based on Autobahn. I will reproduce it and share the logs with you. Thank you for your help

withme8011, 2023-02-03:
I'm not good at English, so I'm writing with a translator.
I'm having the same situation as dkarakay.
[image]
Always get False for the above code.
[image]
This is the code I tested.
The result of my test is that it seems to work well if I connect quickly (within 1 second) after closing, but I have confirmed that it does not work well if I give a delay of more than 5 seconds.
I want to solve this problem

withme8011, 2023-02-08:
I share my test results.
It was confirmed that the test code of gonzalocasas works perfectly on Windows 10.
I think the OS or python version is the problem, so I tested it alternately on Windows 10/CentOS7.9 / Ubuntu22.04, python 3.8.16 / 3.10.2, and it was confirmed that it works abnormally only in Linux.
At this time, SYN did not appear in wireshark.
I think there might be a problem with the scheduling of the function that connects after close internally. Because the SYN doesn't show up in wireshark, it actually indicates that the connect function isn't being called.
If I have any additional confirmed results, I will share them.