RobotWebTools/roslibpy #101: Doesn't work with ROS2 Humble

Ankur-singh, 2022-11-23:
**Description**
For my research project, I am trying to using `roslibpy` with ROS2 Humble. I followed [this doc](https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debians.html) to install ROS2. I am able to run the `listener` and `talker` nodes without any error. 

Then I created [`listener.py`](https://roslibpy.readthedocs.io/en/latest/examples.html#writing-the-listener-node) and [`talker.py`](https://roslibpy.readthedocs.io/en/latest/examples.html#writing-the-talker-node) by following the example from `roslibpy`. I am getting multiple errors in python3. I think, the package and its dependencies are meant to work with python2 only.

On the ROS side, I tried installing and launching the `rosbridge_suite` as suggested [here](https://roslibpy.readthedocs.io/en/latest/reference/index.html#ros-setup-1). **Note:** I replaced "kinetic" with "humble".  But I am getting errors like `Unable to locate: ros-humble-tf2-web-republisher`. And also, I am not able to launch `rosbridge_server`.

Any help would be greatly appreciated.

**System:**
 - OS: Ubuntu 22.04
 - Python version: 3.8, 3.10
 - Python package manager: pip

gonzalocasas, 2022-11-23:
> On the ROS side, I tried installing and launching the rosbridge_suite as suggested [here](https://roslibpy.readthedocs.io/en/latest/reference/index.html#ros-setup-1). Note: I replaced "kinetic" with [...]

The web republisher might not be that important, but without the `rosbridge_server`, it is impossible to connect to ROS with `roslibpy`. Please make sure that you have that running first, and then update this issue with your findings. I believe after you solve that, this issue will be gone.

Ankur-singh, 2022-11-25:
Thank you @gonzalocasas for the prompt response. I tried install `rosbridge_server` using the following command
```bash 
sudo apt install ros-humble-rosbridge-server
```
After a few attempts it ran successfully. Then I resourced ROS and tried launching it using 
```bash
ros2 launch rosbridge_server rosbridge_websocket.launch
```

But I got the following error
```bash
file 'rosbridge_websocket.launch' was not found in the share directory of package 'rosbridge_server' which is at '/opt/ros/humble/share/rosbridge_server'
```

I am new to ROS, so I am not sure what it means. I tried exploring the '/opt/ros/humble/share/rosbridge_server/` directory. I can see a `launch` directory with `rosbridge_websocket_launch.xml` file. I don't know what's wrong and what else to look for.

koichirokato, 2023-01-07:
When executing ROS launch, the file name should be specified directly. The launch file name has changed between ROS1 and ROS2.
If you run below command, you can start rosbridge server.

```
ros2 launch rosbridge_server rosbridge_websocket_launch.xml
```

gonzalocasas, 2023-10-31:
@Ankur-singh if I understand this right, this one has been solved by starting the ros bridge using the command above, right?

ChiaChing-Yen, 2024-05-08:
@gonzalocasas 

I encountered the same issue when following some ROS 2 tutorials online.

> When executing ROS launch, the file name should be specified directly. The launch file name has changed between ROS1 and ROS2. If you run below command, you can start rosbridge server.
> 
> ```
> ros2 launch rosbridge_server rosbridge_websocket_launch.xml
> ```

koichirokato's solution worked for me. This issue occurred due to incorrect syntax when launching the rosbridge-server.
I tested the listener and talker nodes from the examples in the documentation, and they worked smoothly.

My setup:

- ROS 2 Humble on Ubuntu 22.04
- roslibpy 1.7.0

gonzalocasas, 2024-05-08:
Great, thanks! @ChiaChing-Yen That means we can close this, right?

ChiaChing-Yen, 2024-05-08:
I encountered the same issue with @Ankur-singh, but I was able to resolve it by following @koichirokato's suggestion. No modifications were made to the `roslibpy` package.

Given that the solution worked, we can consider closing this issue.