RobotWebTools/roslibpy #109: Header format changed in ROS2

tlpss, 2023-02-03:
https://github.com/gramaziokohler/roslibpy/blob/db741a630880bc5fa33db5e66b44dc6bd032be5b/src/roslibpy/core.py#L55

tlpss, 2023-02-03:
and this would also require dealing with the new class [here](https://github.com/gramaziokohler/roslibpy/blob/db741a630880bc5fa33db5e66b44dc6bd032be5b/src/roslibpy/core.py#L134)

gonzalocasas, 2023-02-03:
Thanks for the report!

LeisAlAyoubi, 2023-08-04:
> The sequence ID has apparently been [dropped in ROS2](https://docs.ros2.org/foxy/api/std_msgs/msg/Header.html) so if you try to use the Header class you get an error: [image]
> 
> https://github.com/gramaziokohler/roslibpy/blob/db741a630880bc5fa33db5e66b44dc6bd032be5b/src/roslibpy/core.py#L55
> 
> Possible solutions include a flag to switch between ROS1 and ROS2 or adding a separate ROS2 Header class:
> 
> ```
> class ROS2Header(UserDict):
>     """Represents a message header of the ROS 2 type std_msgs/Header."""
> 
>     def __init__(self, stamp=None, frame_id=None):
>         self.data = {}
>         self.data["stamp"] = Time(stamp["secs"], stamp["nsecs"]) if stamp else None
>         self.data["frame_id"] = frame_id
> ```

I'm using your ROS2Header-Class but when Im echoing the topic and looking at the message, It seems that the frame_id is not being populated with the string I put in. It works fine with the stamp tho. Is this an issue for you too?