RobotWebTools/roslibpy #127: [Bug] Cannot connect to ROS Master - HTTP/1.0 error

BlackRose01, 2025-02-04:
**Description**
Hi,

I try to connect to a MiR 250 AGV but when I start the script then I get the following error: `WebSocket connection closed: Code=1006, Reason=connection was closed uncleanly ("Unsupported HTTP version "HTTP/1.0"")`. It is possible to connect to the ROS interface with the ROS package from ROS itself and rospy.

I saw that this library is based on the Python package "Autobahn" and I found this 3 lines of code where the error message could come from:
* [https://github.com/crossbario/autobahn-python/blob/7bc85b34e200640ab98a41cfddb38267f39bc92e/autobahn/websocket/protocol.py#L3547](https://github.com/crossbario/autobahn-python/blob/7bc85b34e200640ab98a41cfddb38267f39bc92e/autobahn/websocket/protocol.py#L3547)
* [https://github.com/crossbario/autobahn-python/blob/7bc85b34e200640ab98a41cfddb38267f39bc92e/autobahn/websocket/protocol.py#L2643](https://github.com/crossbario/autobahn-python/blob/7bc85b34e200640ab98a41cfddb38267f39bc92e/autobahn/websocket/protocol.py#L2643)
* [https://github.com/crossbario/autobahn-python/blob/7bc85b34e200640ab98a41cfddb38267f39bc92e/autobahn/websocket/protocol.py#L3752](https://github.com/crossbario/autobahn-python/blob/7bc85b34e200640ab98a41cfddb38267f39bc92e/autobahn/websocket/protocol.py#L3752)

If you need further information let me know. I hope we can fix that problem.

KR,
BlackRose01

*List with ROS version on MiR 250:*
```txt
rosbag_storage: 1.17.0
rosparam: 1.17.0
roscreate: 1.15.8
roslz4: 1.17.0
geometry_msgs: 1.13.1
rosout: 1.17.0
actionlib_msgs: 1.13.1
roslaunch: 1.17.0
rosbridge_library: 0.11.17
message_generation: 0.4.1
diagnostic_msgs: 1.13.1
bondcpp: 1.8.6
roslang: 1.15.8
bondpy: 1.8.6
nodelet_topic_tools: 1.11.0
sensor_msgs: 1.13.1
rosgraph_msgs: 1.11.3
rosclean: 1.15.8
rostime: 0.7.3
rosbuild: 1.15.8
roscpp_traits: 0.7.3
geneus: 3.0.0
rosbag_migration_rule: 1.0.1
gencpp: 0.7.0
roslisp: 1.9.25
mk: 1.15.8
pluginlib: 1.13.0
genlisp: 0.4.18
shape_msgs: 1.13.1
roswtf: 1.17.0
genpy: 0.6.15
stereo_msgs: 1.13.1
rospack: 2.6.2
rosmsg: 1.17.0
rosnode: 1.17.0
rosbash: 1.15.8
rosconsole: 1.14.3
class_loader: 0.5.0
rosbridge_server: 0.11.17
nav_msgs: 1.13.1
rosbridge_msgs: 0.11.17
trajectory_msgs: 1.13.1
message_filters: 1.17.0
rosauth: 1.0.1
bond: 1.8.6
rosbag: 1.17.0
gennodejs: 2.0.2
visualization_msgs: 1.13.1
rosunit: 1.15.8
rosboost_cfg: 1.15.8
std_msgs: 0.5.13
genmsg: 0.6.0
roscpp_serialization: 0.7.3
message_runtime: 0.4.13
std_srvs: 1.11.3
rosmake: 1.15.8
rospy: 1.17.0
dynamic_reconfigure: 1.7.3
rosgraph: 1.17.0
rosservice: 1.17.0
rosmaster: 1.17.0
cmake_modules: 0.5.0
rostopic: 1.17.0
catkin: 0.8.10
rosapi: 0.11.17
nodelet: 1.11.0
actionlib: 1.14.0
smclib: 1.8.6
rostest: 1.17.0
ros_environment: 1.3.2
cpp_common: 0.7.3
xmlrpcpp: 1.17.0
roscpp: 1.17.0
topic_tools: 1.17.0
rosconsole_bridge: 0.5.4
roslib: 1.15.8
```

**To Reproduce**
Steps to reproduce the behavior:
1. Run an Websocket HTTP/1.0
2. Install roslibpy - latest version
3. Save the json file and the python script
4. Run the script
5. See error

**Expected behavior**
A successul connection

**Screenshots**
I hope code is also fine

*topics2.json*
```json
{
    "sinks": {
        "mqtt": {
            "host": "mqtt",
            "port": 1883,
            "topic_prefix": "mir_250",
            "version": "5",
            "transport": "tcp"
        },
        "rtmp": {
            "proto": "rtmp",
            "host": "rtmp_server",
            "port": 1935
        }
    },
    "source": {
        "host": "192.168.1.180",
        "port": 11311,
        "secured": false
    },
    "topics": [
        {
            "ros_topic": "/scan",
            "parameters": {
                "mqtt_alt_topic": "",
                "retain": false
            },
            "sinks": [
                "mqtt"
            ]
        }
    ]
}
```

*Python script*
```python
import json
from roslibpy import Ros
import logging

logger = logging.getLogger(__name__)
logging.basicConfig(
    level=logging.DEBUG
)

def main():
  # Load JSON file
    with open("./topics2.json") as f:
        config = json.load(f)
  
    # Connect to ROS Master
    client = None
    subscriptions = []
    try:
        client = Ros(host=config["source"]["host"], port=int(config["source"]["port"]), is_secure=config["source"]["secured"])
    except:
        try:
            client = Ros(host=config["source"]["host"], port=int(config["source"]["port"]), is_secure=config["source"]["secured"])
        except:
            logging.error("No connection possible")
            import sys
            sys.exit()
    finally:
        client.run()

if __name__ == "__main__":
    main()
```

*Log output*
```log
DEBUG:roslibpy:Started to connect...
INFO:twisted:failing WebSocket opening handshake ('Unsupported HTTP version "HTTP/1.0"')
WARNING:twisted:dropping connection to peer tcp4:192.168.1.180:11311 with abort=True: Unsupported HTTP version "HTTP/1.0"
INFO:roslibpy:WebSocket connection closed: Code=1006, Reason=connection was closed uncleanly ("Unsupported HTTP version "HTTP/1.0"")
DEBUG:roslibpy:Lost connection. Reason: [Failure instance: Traceback (failure with no frames): <class 'twisted.internet.error.ConnectionAborted'>: Connection was aborted locally using ITCPTransport.abortConnection.
]
INFO:twisted:<twisted.internet.tcp.Connector instance at 0x74bbf18d62d0 disconnected IPv4Address(type='TCP', host='192.168.1.180', port=11311)> will retry in 2 seconds
INFO:twisted:Stopping factory <roslibpy.comm.comm_autobahn.AutobahnRosBridgeClientFactory object at 0x74bbf18d6150>
INFO:twisted:Starting factory <roslibpy.comm.comm_autobahn.AutobahnRosBridgeClientFactory object at 0x74bbf18d6150>
DEBUG:roslibpy:Started to connect...
INFO:twisted:failing WebSocket opening handshake ('Unsupported HTTP version "HTTP/1.0"')
WARNING:twisted:dropping connection to peer tcp4:192.168.1.180:11311 with abort=True: Unsupported HTTP version "HTTP/1.0"
INFO:roslibpy:WebSocket connection closed: Code=1006, Reason=connection was closed uncleanly ("Unsupported HTTP version "HTTP/1.0"")
DEBUG:roslibpy:Lost connection. Reason: [Failure instance: Traceback (failure with no frames): <class 'twisted.internet.error.ConnectionAborted'>: Connection was aborted locally using ITCPTransport.abortConnection.
]
INFO:twisted:<twisted.internet.tcp.Connector instance at 0x74bbf18d62d0 disconnected IPv4Address(type='TCP', host='192.168.1.180', port=11311)> will retry in 7 seconds
INFO:twisted:Stopping factory <roslibpy.comm.comm_autobahn.AutobahnRosBridgeClientFactory object at 0x74bbf18d6150>
Traceback (most recent call last):
  File "/usr/src/app/app.py", line 42, in <module>
    client.run()
  File "/usr/local/lib/python3.12/site-packages/roslibpy/ros.py", line 106, in run
    raise RosTimeoutError("Failed to connect to ROS")
roslibpy.core.RosTimeoutError: Failed to connect to ROS
```

**System (please complete the following information):**
 - OS: `Docker - python:3.12-slim`
 - Python version `3.12`
 - Python package manager `pip`
 - Installed Packages (all latest version): opencv, numpy, roslibpy, paho-mqtt

**Additional context**

gonzalocasas, 2025-02-04:
I don't understand the setup. Why are you trying to connect to port `11311`? That's not the ROS Bridge port usually. Could it be that you're trying to connect to an unrelated/custom websockets server running on that port?

gonzalocasas, 2025-02-04:
For clarification, the host that you connect to (`192.168.1.180` according to your config), should be a computer running `rosbridge server` (it can be the same a ROS core or a node on the ROS network, that's irrelevant).

BlackRose01, 2025-02-05:
Hey,

sorry for that. I tried now for 3 days to get that working and I always received the error but I saw today that I missed to try that Host/Port combination. The "problem" is that the MiR robot has 2 to 3 different internal networks but only 1 Hotspot to connect to. So it worked now. Netherless, thanks for the help. 👍 

KR,
BlackRose01

closed 2025-02-05 by BlackRose01