RobotWebTools/roslibpy #128: Way to get topic publishers and subscribers?

diasdm, 2025-04-11:
Hi!

Is there a way to get the list of publishers and subscribers of a given topic? Something like we see with `rostopic info`

Thank you!

gonzalocasas, 2025-04-11:
There is some access to this type of information, check the docs here: https://roslibpy.readthedocs.io/en/latest/examples.html#querying-ros-api

Is that what you are referring to?

diasdm, 2025-04-11:
From what I can see none of those allow to get a list of the nodes which are subscribed / publishing a given topic

I took a deeper look and it seems the services exist, see working example:
```
from roslibpy import Ros, Service, ServiceRequest
from roslibpy.ros import ROSAPI_TIMEOUT

class MyRos(Ros):
    def get_topic_subscribers(self, topic: str, callback=None, errback=None):
        service = Service(self, "/rosapi/subscribers", "rosapi/Subscribers")

        result = service.call(ServiceRequest({"topic": topic}), callback, errback, timeout=ROSAPI_TIMEOUT)

        if callback:
            return

        return result["subscribers"]

    def get_topic_publishers(self, topic: str, callback=None, errback=None):
        service = Service(self, "/rosapi/publishers", "rosapi/Publishers")

        result = service.call(ServiceRequest({"topic": topic}), callback, errback, timeout=ROSAPI_TIMEOUT)

        if callback:
            return

        return result["publishers"]

client = MyRos(host='localhost', port=9090)
client.run()

print(client.get_topic_subscribers("/rosout"))
print(client.get_topic_publishers("/rosout"))

client.terminate()
```